microsoft/AirSim #717: Low fps with many cameras

anders-hopland, 2018-01-02:
I am building a simulator which is using 5 cameras. TThe framerate is around 30fps while rendering to screen, but when I´m sending it to the decision making engine, I can only get out 4fps. I think the bottleneck is moving data from GPU to CPU memory. Have someone encountered a similar problem, or have any ideas on how to solve this. 

Thanks for your help!

anders-hopland, 2018-01-06:
I'm using a GTX1070 on windows in a small environment. The cameras are 720p each.

anders-hopland, 2018-01-08:
Are you referring to the Airsim API or the in game FPS? Do you know about any "hacky" solutions?

sytelus, 2018-01-13:
I would suggest first try out 5 cameras with lower resolution (default). If that doesn't work then you have bigger issue. Then try to reduce cameras from 5 to 4 to 3. See which config gives you reasonable FPS. I'm able to use 3 cameras comfortably on TitanX in Blocks environment at default res.

sytelus, 2018-02-13:
We have some good proposals here: https://github.com/Microsoft/AirSim/issues/796

We'll continue discussion in above thread.

closed 2018-02-13 by sytelus