microsoft/AirSim #718: Prebuild Binaries Segmentation

moritzlbr, 2018-01-03:
My issue is similiar to this question: https://github.com/Microsoft/AirSim/issues/702

I'm trying to get a set of two images ("scene" and "segmentation") from the city environment binaries v1.1.7. I'm using Windows 7 Enterprise and Unreal Engine 4.16.3.
However the segmentation won't change to different colors.

Here's what I get:
[image]
[image]

I tried using the following code though nothing changed:
_client.simSetSegmentationObjectID("[\w]*", 0, True)
client.simSetSegmentationObjectID("ground[\w]*", 1, True)
client.simSetSegmentationObjectID("road[\w]*", 2, True)
client.simSetSegmentationObjectID("sky[\w]*", 3, True)
client.simSetSegmentationObjectID("car[\w]*", 4, True)
client.simSetSegmentationObjectID("sign[\w]*", 5, True)_

I'm not sure if I got the names of all the meshes' stencils right, that might already be the problem.
Could you publish the stencils' names for the released environments?
Or do you have any idea what else could have gone wrong?

Thanks for your help!

sytelus, 2018-01-06:
In current beta we had to turn off setting initial object IDs in city environment because of perf issue (the environment is very large). I'm looking in to a workaround and will enable it again soon...

moritzlbr, 2018-01-09:
Thanks for your answer!
Than I'll patiently wait until you've found a solution.

closed 2018-01-09 by moritzlbr