microsoft/AirSim #720: Can't run release program normally.

Qinja, 2018-01-05:
Platform: Windows 10 x64 . 
Version: Release V1.17 LandscapeMountains
Issue: I extracted zip file and run the exe file (and bat file), but it stopped working with a message box "Landscape Mountains Showcase has stopped working“.
I use 4 computers to run it but received a same result.

sytelus, 2018-01-06:
Just downloaded same zip and tried it out. It seems to work as expected on my m/c. Could you see if there are any logs created in below location?

LandscapeMountains\Saved\Logs

Are you using English version of Windows? Any special settings in OS? Do you have GPU?

Qinja, 2018-01-06:
Thanks a lot. 
Logs folder does not exist. 
The OS is Win10 x64 Chinese version. 
GPU is GTX1070.
I haved tested it in 5 computer. 1 is  successful  , 4 are error with the same result.

I found a folder named Crashes .
[Crashes.zip](https://github.com/Microsoft/AirSim/files/1608592/Crashes.zip)

sytelus, 2018-01-10:
Inside Crashes folder there is a file called CrashContext.runtime-xml which has following line:

`<ErrorMessage>Fatal error: [File:D:\Build\++UE4+Release-4.16+Compile\Sync\Engine\Source\Runtime\PakFile\Private\IPlatformFilePak.cpp] [Line: 3621] &nl;Trailing magic number (0d) in &apos;../../../LandscapeMountains/Content/Paks/._LandscapeMountains-WindowsNoEditor.pak&apos; is different than the expected one. Verify your installation.&nl;</ErrorMessage>`

I don't know what means... May be your download zip file is corrupted? So downloading again and try might help. Also, you can try posting at https://answers.unrealengine.com/.

Qinja, 2018-01-10:
Thank you!
I tried downloading again but failed.
I tried building a blank scene and it worked well.
Probably as you said, it's some error with unreal but not airsim.
Thank you!

closed 2018-01-10 by Qinja