microsoft/AirSim #721: How can I make a x86 project?

Qinja, 2018-01-06:
I had built project in x64 and it work well. I considered a 32bit version, then I had tried the following steps:

1.  I made rpc.lib (win32) and built Project AirSim.sln (win32) successfully. 

2. Then I copied **\AirLib**  to **\Unreal\Plugins\AirSim\Source** and copied  **Mavlinkcom**(win32) to **\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\lib** to Update **AirLib** and **Mavlinkcom** as "build.cmd" file do.

3. And I copied whole **Plugins** floder to my unreal project root as I had done in x64.

4. I run my unreal editor and packed project for windows (32 bit) , unreal output message **BUILD FAILED** and the output log showed: 
> ...\Plugins\AirSim\Source\AirLib\src\api\RpcLibClientBase.cpp(22): fatal error C1083: Can't open include file rpc/client.h: No such file or directory
> ERROR: UBT ERROR: Failed to produce item: ...\Plugins\Airsim\Binaries\Win32\UE4-AirSim-Win32-DebugGame.lib

5.  I confirmed the files **rpc/client.h** is exist, then I opened unreal VS project and generate my unreal vc++ project(win32), received the same error **C1083**
> Can't open include file rpc/client.h: No such file or directory

6. I added include directory and lib directory to VC++ directory and rebuilt project. Error C1083 solved. but **LNK 2019** occured. seem that lib files didn't link properly.
>Unresolved external symbols XXXX

7. I checked my process but I didn't know how to solve. I need some help. Thanks!

sytelus, 2018-01-13:
Currently we are not supporting x86 compilation (only x64). Is there any reason you need x86 binaries? I suppose this is doable but will need some effort. First thing would be to compile rpclib for x86 and then make sure VS.Net uses that (using Build Configuration). Then make sure AirSim.Build.cs is updated to use x86 binaries. I think that should do it. Please feel free to contribute if this is important for your scenario.

Qinja, 2018-01-13:
Thank you. I had made rpclib and mavlinkcom for x86 but I didn't link them correctly.
I wanted to make x86 binaries to test issue #720 . It is now solved.
Maybe I will keep trying making a x86 project. 
Thank you .

closed 2018-01-13 by Qinja