microsoft/AirSim #723: update_from_git.sh fails.

ssashir06, 2018-01-08:
The shell script update_from_git.sh in Blocks fails because it says "No such file or directory".
I think rsync has wrong arguments, and fixing it works fine.

The environment I have checked is ubuntu 16.04.  How is it working in another environment?
In addition, on the macOS, rsync creates the missing directory automatically and copies some files in wrong location.

sytelus, 2018-01-09:
What is the fix you made?

ssashir06, 2018-01-13:
Currently, the DEST argument of rsync has "AirSim" path. And also, the SOURCE argument has "AirSim" too. This will makes being copied file will be placed in "AirSim/AirSim" if this path exists.

Could you please see a new pull request I created?
Thank you.

PR #730 (2018-01-18): fix rsync arguments
https://github.com/Microsoft/AirSim/issues/723

fix rsync arguments to make copy files correctly to resolve #723
closed 2018-01-18 by sytelus