microsoft/AirSim #724: ERROR - Retry connection over the limit for hello_car.py

DQQ-077-SZ, 2018-01-08:
When I run the hello_car.py, I got the ERROR "Retry connection over the limit". How can I solve it? Unfortunately, I could not google a proper answer to solve it. Thanks for your help.
[image]
[image]

sytelus, 2018-01-09:
Are you running Unreal environment like Blocks in other window? Please let us know which docs you are following.

DQQ-077-SZ, 2018-01-10:
Thanks for your help @sytelus . I should run the Blocks environment before running the Python code. Now I can run it. Thanks for your time again.

[image]

closed 2018-01-12 by sytelus