microsoft/AirSim #725: memoryerror appeared when running the DQNcar.py

Mikemraz, 2018-01-08:
when opening the neighborhood environment and running AirSim/PythonClient/DQNcar.py, a memoryerror appeared:
[image]

my python's version is 
[image]

Can anyone help me out?

sytelus, 2018-01-13:
There was a runtime error in DQNcar.py because of interface change which I just fixed in reference commit. After this fix, I'm not seeing any memory issues. Do you encounter above issue right away or after sometime? In any case, please try out new version.

closed 2018-01-13 by sytelus

yujianyuanhaha, 2018-11-09:
I met the error again.  

``` bash
/home/jet/anaconda3/envs/py2/lib/python2.7/site-packages/cntk/cntk_py_init.py:47: UserWarning: Unsupported Linux distribution (ubuntu-18.04). CNTK supports Ubuntu 16.04 and above, only.
  warnings.warn('Unsupported Linux distribution (%s-%s). CNTK supports Ubuntu 16.04 and above, only.' % (__my_distro__, __my_distro_ver__))
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "DQNcar.py", line 505, in <module>
    agent = DeepQAgent((NumBufferFrames, SizeRows, SizeCols), NumActions, monitor=True)
  File "DQNcar.py", line 270, in __init__
    self._memory = ReplayMemory(memory_size, input_shape[1:], 4)
  File "DQNcar.py", line 35, in __init__
    self._states = np.zeros((size,) + sample_shape, dtype=np.float32)
MemoryError
```

Curry30h, 2019-03-21:
I also encountered the same problem. Is there a solution?