microsoft/AirSim #729: How can I change the speed of drones in Computer vision mode?

saihv, 2018-01-11:
In Computer vision mode, the drones can be moved with the same keys as in the manual camera mode: how can I change the speed of this movement?

saihv, 2018-01-12:
Figured it out, I needed to change the scaling factor (default value 1000) on [line 116 of ManualPoseController.cpp](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/ManualPoseController.cpp#L116)

closed 2018-01-12 by saihv

chickenbestlover, 2020-11-16:
That's exactly what I was looking for. Thank you very much!

rajrana22, 2021-07-06:
When I change the speed_scaler value, nothing changes. I'm still moving at the same speed (which is too fast) no matter what I change or how much I change it by.

rajrana22, 2021-07-07:
What exactly is the "speed_scaler" value measured in? I know that AirSim uses SI units and Unreal Engine uses cm, but neither of those would make sense as values (too large). Is it meters per second? Is it cm per second? Or is it not a speed but an absolute distance moved with each press of the forward button?

jonyMarino, 2021-07-07:
answered here https://github.com/microsoft/AirSim/issues/3783#issuecomment-875938514