microsoft/AirSim #734: MavLink (PX4 sitl) triggering camera

kjetilkjeka, 2018-01-14:
We're doing an application that requires the possibility of planning a mission from QGC with camera caputre. It seems like AirSim currently doesn't do anything with the corresponding MavLink message, would it be of interest to have such features up-streamed?

If my limited understanding of the codebase is correct it should be implemented in the `handleMessage` method of a subclass of `MavLinkNodeImpl`. Would [`MavLinkVehicle`](https://github.com/Microsoft/AirSim/blob/4a7fe72cdec6a2651e5b578a7d6f51a1caacf2e6/MavLinkCom/src/impl/MavLinkVehicleImpl.cpp) or a new class similar to [`MavLinkVideoServerImpl`](https://github.com/Microsoft/AirSim/blob/4a7fe72cdec6a2651e5b578a7d6f51a1caacf2e6/MavLinkCom/src/impl/MavLinkVideoStreamImpl.cpp) make the most sense?

sytelus, 2018-01-18:
Which messages you are interested in handling? Currently we only handle actuator messages, i.e., how vehicle moves around.

kjetilkjeka, 2018-01-18:
Since the HITL messages is implemented, and it's reasonable to use AUX channels for the gimbal the only new message i would like to implement is the [CAMERA_TRIGGER](http://mavlink.org/messages/common#CAMERA_TRIGGER) message.

The [PX4 tutorial for camera trigger](https://dev.px4.io/en/tutorials/camera_trigger.html) states that if `TRIG_INTERFACE = 3` the autopilot will send a `CAMERA_TRIGGER` over MavLink. This is the message i wish to react to. The desired behavior would be similar to starting and stoping a recording except exactly one frame would be stored per `CAMERA_TRIGGER` message.

kjetilkjeka, 2018-01-21:
I've made this work but I'm no longer sure that using the `CAMERA_TRIGGER` message is the correct one. I've created an [issue on the MavLink repo](https://github.com/mavlink/mavlink/issues/828) to hopefully get the correct message to use.

After i get a response from the MavLink guys, [support in PX4](https://github.com/PX4/Firmware/issues/8739) is the next step.

sytelus, 2018-02-13:
Let us know how this goes. We haven't used these features yet... It would be great if you can submit a PR with your change!

sytelus, 2018-02-21:
On related note, I just added Gimbal APIs in AirSim: https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#camera-orientation-gimble