microsoft/AirSim #735: Error at startup: No such file or directory: unspecified iostream_category error

mchen0037, 2018-01-15:
Hi, I was trying to run the [Blocks simulation](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md), but I ran into this error: 
[image]
I was following the instructions under Linux (16.04) and was prompted for incompatible version and conversion (step 3) but was not prompted about missing modules nor building with AirSim (5).

I rebuilt both Airsim and UE4Editor (4.17.2) from scratch, but I got the same error. Any suggestions?

Thanks!

Mikhail2017, 2018-01-15:
Same problem

mchen0037, 2018-01-15:
@Mikhail2017 I built the v1.1.6 version for AirSim and it worked. 
[image]

tejaswid, 2018-01-16:
If you add the following lines to the settings.json file (just replace car with Multirotor in the car example), the rotor launches fine. But once you press Esc in play mode the simulation crashes.
```
{
  "SettingsVersion": 1.0,
  "SimMode": "Multirotor"
}

Tested on 4.18.2 in Ubuntu 16.04.
```

sytelus, 2018-01-20:
I found the issue and its now fixed in master branch. The problem was occuring if you already did not have settings.json.

closed 2018-01-20 by sytelus

olivia-skydio, 2018-01-20:
I just pulled from master and tried to rebuild and got this error:

/home/olivia/AirSim/DroneServer/main.cpp:71:79: error: no member named 'getFileName' in 'msr::airlib::Settings'; did you mean 'hasFileName'?
        std::cout << "Could not load settings from " << Settings::singleton().getFileName() << std::endl;
                                                                              ^~~~~~~~~~~
                                                                              hasFileName
/home/olivia/AirSim/AirLib/include/common/Settings.hpp:101:10: note: 'hasFileName' declared here
    bool hasFileName()
         ^
6 warnings and 1 error generated.
DroneServer/CMakeFiles/DroneServer.dir/build.make:62: recipe for target 'DroneServer/CMakeFiles/DroneServer.dir/home/olivia/AirSim/DroneServer/main.cpp.o' failed
make[2]: *** [DroneServer/CMakeFiles/DroneServer.dir/home/olivia/AirSim/DroneServer/main.cpp.o] Error 1
CMakeFiles/Makefile2:556: recipe for target 'DroneServer/CMakeFiles/DroneServer.dir/all' failed
make[1]: *** [DroneServer/CMakeFiles/DroneServer.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

I had just downloaded AirSim and just built recently and had no problem before I pulled.

asimonov, 2018-01-26:
I pulled today the current master and seen this issue on Mac with 4.17.2 (pulled from Epic). where should I find settings.json?