microsoft/AirSim #736: LINK : fatal error LNK1104: cannot open file "rpc.lib"

chmod4755, 2018-01-15:
When I want to run the 'Hello Car' project to control my car in released environment(binaries) , I couldn't build this 'Hello car' project.
Is it means I must build AirSim first?

sytelus, 2018-01-18:
Please look at the docs for how to build: https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md

closed 2018-02-13 by sytelus

LarsPede, 2018-04-11:
Hi @sytelus 
I have kinda the same problem.
I am following the docs: 
https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md
to the point.
I have no problem running build.cmd and I have no issue running the exe files from the debug folders, but I get the following output 
`1>Project not selected to build for this solution configuration 
2>------ Skipped Rebuild All: Project: AirLib, Configuration: Debug Win32 ------
2>Project not selected to build for this solution configuration 
3>------ Rebuild All started: Project: HelloCar, Configuration: Debug Win32 ------
3>  main.cpp
3>LINK : fatal error LNK1104: cannot open file 'rpc.lib'
========== Rebuild All: 0 succeeded, 1 failed, 2 skipped ==========`
when I try to build the "HelloCar" project. This, of course, results in the error:
`Error	LNK1104	cannot open file 'rpc.lib'	HelloCar	C:\Users\{username}\source\repos\CarSim15\HelloCar\LINK	1`

To clarify my current build process and specs:

I am using:
VS15 With Update 3 with default build.cmd - but had the same issue on VS17 (after correcting the build.cmd).
Latest version of CMake.

Build.cmd:
Has completed successfully with no errors.