microsoft/AirSim #742: Recommendation for steering wheel

deanzadok, 2018-01-17:
Hi everyone,

We want to buy a steering wheel to record dataset from AirSim.
Do you have any recommendations for one that will work better with AirSim (just like in the gif in the readme)?

Thanks,

Dean.

sytelus, 2018-01-18:
I've tested with Logitech G920 steering and pads and AirSim defaults are currently setup to use that wheel. 

However it's worth noting that I wasn't too thrilled with the realism of this wheel and pads. Perhaps more tuning might make this better for your scenario. This tuning can be done at [these lines](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Car/CarPawn.cpp#L429) of code:

```
        joystick_controls_.steering = joystick_state_.left_y * 1.25;
        joystick_controls_.throttle = (-joystick_state_.right_x + 1) / 2;
        joystick_controls_.brake = -joystick_state_.right_z + 1;
```

So you can customize how responsive your steering is by setting the right constant, for example.

deanzadok, 2018-01-18:
Hi Shital,

Thanks a lot! 
We will probably go for the g920 with tuning these parameters.

closed 2018-01-18 by deanzadok