microsoft/AirSim #744: Error at startup: Invalid argument: iostream stream error

AemaH, 2018-01-18:
Hi, I try to create Unreal Environment like the [tutorial doc](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md), but I ran into this error.[image]
OS is win7, and I use the VS2015 update

mchen0037, 2018-01-18:
Did you try some of the suggestions on https://github.com/Microsoft/AirSim/issues/735 ?

sytelus, 2018-01-20:
I found the issue and its now [fixed](https://github.com/Microsoft/AirSim/commit/065ad2b3312b78350ec252d6842e5d6748e1e7d0) in master branch. The problem was occuring if you already did not have settings.json.

closed 2018-01-20 by sytelus

asimonov, 2018-01-26:
I have this latest change but still see the error. where settings.json should be? and how is it created? I cannot seem to find it

SamHS-OCI, 2018-03-15:
I am having the same error, but when I try to run from the provided binaries. What am I doing wrong?