microsoft/AirSim #745: PointCloud generation

badabook, 2018-01-18:
I tryed to generate a pointcloud using a point_cloud.py update. The modifications are due to the fact that there is no more "Depth" as ImageType in AirSim, here is the code: 

```
while True:
    rawImage = client.simGetImage(0, AirSimImageType.DepthVis)
    if (rawImage is None):
        print("Camera is not returning image, please check airsim for error messages")
        sys.exit(0)
    else:
        png = cv2.imdecode(np.frombuffer(rawImage, np.uint8) , cv2.IMREAD_UNCHANGED)
        gray = cv2.cvtColor(png, cv2.COLOR_BGR2GRAY)
        Image3D = cv2.reprojectImageTo3D(gray, projectionMatrix)
        savePointCloud(Image3D, outputFile)
        print("saved " + outputFile)
        sys.exit(0)

    key = cv2.waitKey(1) & 0xFF;
    if (key == 27 or key == ord('q') or key == ord('x')):
        break;
```
But that's what i obtain:

[image]

This is the link to Sketchfab: https://skfb.ly/6vQ7T

What am I doing wrong?

sytelus, 2018-01-20:
Try to use DepthPlanner and DepthPerspective instead of DepthVis. The DepthPerspective mode generates depth as if single ideal camera will generate depth. Cameras use perspective transform so parallel rays are no longer parallel after transformation. [See here](https://medium.com/retronator-magazine/game-developers-guide-to-graphical-projections-with-video-game-examples-part-1-introduction-aa3d051c137d) for more info. The DepthPlanner mode keeps parallel rays parallel.

Also, please note that point_cloud.py was an external contribution and I personally haven't tested it. It would be great if you had like to work on this script, make it better and contribute improved version :). I've already updated this script to use the latest APIs.

badabook, 2018-01-20:
Thank you for the answer! I will work on this script for a new update!

kshitijgoel007, 2018-03-28:
@badabook @sytelus Were you able to register point cloud correctly? I am trying to make a local map using the depth perspective images, but I am getting a radial distortion.

If you go really close to a wall (where the pfm array should be entirely filled with same value), after printing the array it seems it has outward radial distortion.

Here is an example 
[image]
On the right hand side there is the airsim scene from the zen garden environment and on the left hand side is a registration of the scene using depth images. It seems the depth images are radially outward. Did you guys also observe this?

jonyMarino, 2020-09-18:
Closed due to inactivity. Doc update pending

closed 2020-09-18 by jonyMarino