microsoft/AirSim #746: How to achieve data with regular time interval ?

XinGao2017, 2018-01-19:
Hi,
I want to achieve precise 30 FPS or 100 FPS data , so in order to achieve 30 FPS data ,  I set "RecordInterval" to 0.033 , and for 100 FPS , I set "RecordInterval" to 0.01 in settings.json .
After recording , according to the airsim_rec.txt , I find the time interval is not regular .
So I wonder to know how can I achieve data with regular time interval , or what a powerful computer can be used to generate data with regular time interval ?
My computer : 
CPU : Intel Core i7-3610QM 2.30GHz
GPU : NVIDIA GeForce GT 650M
RAM : 8G

sytelus, 2018-01-20:
Are you using quadrotors or cars? You might want to look in to [fixed frame rate](http://www.aclockworkberry.com/unreal-engine-substepping/#Fixed_Frame_Rate) feature in Unreal Engine. Set it to some value that your machine can handle. Overall, it's hard to get very precise intervals however because OSes we are using (Linix/Windows) are themselves not real time. This means you should expect 4-20 ms of variance. Usually its much better idea to design your algorithms and workflow such that it doesn't need very precise timing.

XinGao2017, 2018-01-20:
Hi,
I use the default car, according to your advice, I will need some changes with our algorithms .
Thanks !

closed 2018-02-13 by sytelus