microsoft/AirSim #747: Cannot open file 'opencv_core246.lib' while building MavLinkTest

zbenic, 2018-01-21:
While building AirSim on my system (Win8.1, VS 2015, SP3, CMake 3.10.2) by running the build.cmd I have encountered following error:

`Build FAILED.

"C:\GIT\AirSim\AirSim.sln" (default target) (1) ->
"C:\GIT\AirSim\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj" (default target) (7)
 ->
(Link target) ->
  LINK : fatal error LNK1104: cannot open file 'opencv_core246.lib' [C:\GIT\Air
Sim\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]

    0 Warning(s)
    1 Error(s)`

The needed file exists on the path which is : C:\opencv\build\x64\vc10\lib\opencv_core246.lib.

Can anybody assist me?

sytelus, 2018-02-13:
MavLinkTest doesn't have any dependency on opencv so not sure where this is coming from. Also, I can't reproduce this and none of other users have reported it. You might want to make sure you don't have any changes of your own. Also, turn on verbose compile in Visual Studio (search internet) to see additional messages that might be helpful.

zbenic, 2018-02-13:
Thanks for the answer.
In the meantime I have migrated to Win10.
Had no problems with running the build.cmd there.

closed 2018-02-13 by sytelus