microsoft/AirSim #748: Cannot compile latest as well as v1.1.7 release on windows

SweetShot, 2018-01-22:
Hello, 
I am pretty sure its some small thing but i am constantly getting this specific error while trying to build rpclib.
[image]

Thanks a lot

sytelus, 2018-01-22:
I'm thinking you have some sort of mixed code from old and new release. Last week we removed rpclib as submodule and instead we now download it from GitHub releases as part of the build. I've tested this on couple machines and it should be working fine. I would suggest, try running g`it clean -fdx` followed by `git pull` and then `build`.

SweetShot, 2018-01-24:
Hello,
I re-downloaded entire project now and tried again now, i am getting another error now, I have VS17 installed, will that be the culprit ? should i install VS15 too ?
[image]
Thank you

sytelus, 2018-01-26:
Could you run `git clean -fdx`, `clean.bat` followed by `build.bat`? As you now have VS2015, please make sure to run this from VS2015 x64 Native Tools Command Prompt.

SweetShot, 2018-01-27:
Thanks a lot !!
It worked correctly in VS2015 x64 Native Tools Command Prompt. Quite interested in this project :D will have an amazing time with it.
Closing

closed 2018-01-27 by SweetShot