microsoft/AirSim #750: Makefile:252: recipe for target 'CrashReportClient-Linux-Shipping' failed

mashrin, 2018-01-23:
Getting this error while executing "make" command.

OS - Ubuntu 17.10
Running Unreal 4.17

mashrin, 2018-01-23:
Fixed it.
This helped:

$ chown -R user: filepath //mention user and the path to the directory
$ su user
$ make

Do not use 'sudo' or 'root user' while executing make command as it leads to another error where it shows:
"Refusing to run with the root privileges."

Can be closed now.

closed 2018-01-23 by mashrin

pbasson, 2018-07-01:
I can confirm this issue resolves the Unreal Engine make error.

XiongDuan, 2018-07-16:
”I can confirm this issue resolves the Unreal Engine make error.“
============================================
It happened to me, too. Just resolve it by this case.

Curry30h, 2019-06-05:
“”user: filepath //mention user and the path to the directory “ What is the specific content of this part?

AsherYartsevTech, 2019-09-15:
This error still occures @mashrin

hipforth, 2020-07-13:
$ chown -R yue: /home/yue/UnrealEngine/
$ su user
$ make

It's work for me  
ubuntu18.04 && UE4.24

validsyntax1012, 2021-05-02:
i get same issue after commands as well

lahtis, 2021-05-07:
I get a same error when using chown command. What to do?

Using Ubuntu 18.4.5 LTS