microsoft/AirSim #751: fatal error: 'xlocale.h' file not found

mashrin, 2018-01-23:
Came across this error:
ThirdParty/Linux/LibCxx/include/c++/v1/__locale:39:11: fatal error: 'xlocale.h' file not found.

Solved this by-
ln -s /usr/include/locale.h /usr/include/xlocale.h

This is a workaround. xlocale.h is threadsafe version of locale.h
**Would like to know if there's something else that can be done as well.**

Posting the issue along with the workaround here so that it might benefit the community, if someone comes across the same issue.

sytelus, 2018-01-26:
I have never seen this error. What is yout distro and version of Linux? Had you made any changes to AirSim code?

mashrin, 2018-01-26:
I am using Ubuntu 17.10
Also, same error in Ubuntu 16.10

No, I didn't make any changes to the Airsim code.
It just required this header file which wasn't present.
Searched the system for this file, found locale.h rather than xlocale.h
Searched the web to read about both. Found out that xlocale.h is the threadsafe version of locale.h
So, linked this file and now it works.

sytelus, 2018-02-13:
Looks like this is Unreal Engine specific issue: https://answers.unrealengine.com/questions/705076/xlocaleh-not-found-on-linux.html

As this is unrelated to AirSim, I would suggest to continue discussion in above forum.

closed 2018-02-13 by sytelus