microsoft/AirSim #752: PX4 SITL Calibration for AirSim

mrRedDead98, 2018-01-24:
Hey guys, 

I am able to connect the drone from AirSim and QGroundControl on Windows 10 with PX4 SITL on Ubuntu VM but I have some issues with start flying the multicopter drone with xbox-controller in AirSim Blocks-Environment.

1. There are many sensors and parameters in px4 and I don't know how I have to calibrate them for an AirSim, especially for the preflight. I looked in the px4-config but I don't know all these params. For example in QGC the altitude falls from circa 1 to -170 and jumps to 1 and so on.

2. I set the LPE_LAT and LPE_LON to my local coordinates and px4 says it changes the old ones with my new ones but the [commander] set home to other coordinates and in QGC the drone is at wrong position. Which one of the other coordinates do I have to set up so that I can see the drone at my coordinates in QGroundControl?

[image]

sytelus, 2018-02-13:
These are PX4 related issues and you would have better luck getting answers from PX4 forums (either http://discuss.px4.io/ or https://github.com/PX4/Firmware/issues). We would love to know your finds if its useful to AirSim users as well!

closed 2018-02-13 by sytelus

shangjieid, 2018-03-16:
Hi @mrRedDead98 ,
I have trouble connecting Airsim on Win10 with PX4 on Ubuntu VM, could you show your steps to connect them? Thank you very much.

mrRedDead98, 2018-03-16:
Hi @shangjieid ,
if you can use the ping command successfully, you only need to set the right settings in Settings.json:
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0,
  "DefaultVehicleConfig": "Pixhawk",
  "LocalHostIp": "<your-W10-IP>",
  "Pixhawk": {
    "LocalHostIp": "<your-W10-IP>",
    "LogViewerHostIp": "<your-Ubuntu-IP>",
    "LogViewerPort": 14388,
    "LogViewerSendPort": 14389,
    "FirmwareName": "PX4",
    "SitlIp": "<your-Ubuntu-IP>",
    "SitlPort": 14556,
    "UdpIp": "<your-Ubuntu-IP>",
    "UdpPort": 14560,
    "UseSerial": false
  }
}`

If you compiled PX4 successfully and you run with default config file it should connect.

shangjieid, 2018-03-17:
Thank you very very much, kind man. It's very useful.