microsoft/AirSim #755: find: 'cmake version': No such file or directory  Newer AirSim requires cmake verion "   3.   9" but you have "    .    " which is older. Press any key to download and install cmake (make sure to add it in path in install options)

jayjagtap, 2018-01-25:


sytelus, 2018-01-26:
I think cmake is not on path. You can check this by just typing cmake from AirSim folder. If you get `'cmake' is not recognized as an internal or external command` then it means its not path. When you install cmake, it gives you a choice for putting it on path. Unfortunately default is no so you have to make sure this is changed in install screen. If you have already finished installing cmake, just add its bin folder in path (default path on Windows is `C:\Program Files\CMake\bin`)

jayjagtap, 2018-01-26:
Thak you for your reply.
I had selected the option of setting the path while installing and even checked in the environment variables. Its there.

This is the output when i type cmake from airsim folder.

[image]

Actually I have a urgency for drone training. Please show me a way to get out of this.

SweetShot, 2018-01-27:
Not really sure but it is certainly working with cmake 3.10.2, you might want to try installing that, also it is the version mentioned in check_cmake.bat , link https://cmake.org/files/v3.10/cmake-3.10.2-win64-x64.msi

tsbio, 2018-01-30:
I have the same problem! 'cmake was not found in path.'
  installed VS2015 windows 10 unreal 4.18 ,but just can't run build.cmd

jayjagtap, 2018-01-30:
Check out the temporary solution which would work over here.

https://github.com/Microsoft/AirSim/pull/761/commits/cc8a08cbf7af577e0f2be99218c21b544c66954f

sytelus, 2018-01-30:
Did you restarted command prompt after cmake install? Change in Path environment variable doesn't take effect until you restart app.

vincedjango, 2018-09-20:
What is said above is not wrong and has to be done, but it is not an answer to the question asked.

The problem here is not **CMake**, the problem is the **Find** command in this line:
`'cmake --version ^| find "cmake version"'`

In the cmd prompt, type:
```
where Find.exe

Output:
C:\cygwin64\bin\find.exe
C:\Windows\System32\find.exe

```

The find.exe of System32 has to come first. To do that, make sure that _PATH_ of your _System Environment Variable_ has _cygwin64_ paths at the end.

Once that your ouput is like that, it will work.
```
C:\Windows\System32\find.exe
C:\cygwin64\bin\find.exe

```

p.s: Cygwin64 was only used as an example, the second find.exe could come from another source

mayank-technical, 2019-04-17:
Hi 
Thanks @vincedjango  . Great Help 

But its better to try where Cmake in command promt
C:\cygwin64\bin\cmake.exe
C:\Program Files\CMake\bin\cmake.exe

here you can go to Environment Variables and make sure in user and system variable the position of C:\Program Files\CMake\bin\cmake.exe is above C:\cygwin64\bin\cmake.exe

It worked for me.

AnukritiSinghh, 2021-03-16:
I am having the same issue with VS 2019. Can anyone please let me know if they were able to solve it?