microsoft/AirSim #756: How to Creating and Setting Up Unreal Environment in Linux

kdzy, 2018-01-25:
The  official tutorials  only contain  "Creating and Setting Up Unreal Environment" in windows ,the address is  https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md, I want to know How to Creating and Setting Up Unreal Environment in Linux, I have tried several times ,but I still can't how to olve it,could anynoe tell me How can I  create and set Up Unreal Environment in Linux ?

sytelus, 2018-01-26:
Steps are same accept you need to create initial files in Windows first and then copy them over to Linux machine. We are planning to release source for Mountain environment soon for Linux so hopefully this will be more easier:

https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md#note-for-linux-users

kdzy, 2018-01-27:
You mean I should install Unreal engine in Windows first,then follow the steps of 7-11?
[image]

sytelus, 2018-01-30:
Yes, you will need to create Unreal project on Windows machine using steps like above and then copy over files to Linux machine.

kdzy, 2018-03-14:
Hi  Shital, I'm a user of AirSim from China. I have created and set up  Unreal
Environment on windows 10 successfully. But now ,I want to use AirSim on
Ubuntu, the problem is that how can I create my own environment?   If I
copy over the whole folder of "LandscapeMountains" to Ubuntu,it can't work.
The version of Unreal Engine on windows is 4.16, while Ubuntu is 4.17. It
is the reason of the  version of  Unreal Engine? So from which step I
should follow in the tutorials
https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md. I
want to know what should I do to create my own environment on Ubuntu? I am
looking forward to your reply. Thank you a lot!

Parnia, 2018-07-24:
Hello,

I built the "LandscapeMountains" successfully on windows 10. I copied the environment folder to ubuntu 16.04 as it is stated in the docs, but I encounter this error when I try to open it on unreal editor:
`[5/6] Link libUE4Editor-AirSim.so
/usr/bin/ld: cannot find /home/parnia/AirSim/Unreal/Environments/LandscapeMountains/Plugins/AirSim/Source/AirLib/deps/MavLinkCom/lib/libMavLinkCom.a: No such file or directory
/usr/bin/ld: cannot find /home/parnia/AirSim/Unreal/Environments/LandscapeMountains/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a: No such file or directory
clang: error: linker command failed with exit code 1 (use -v to see invocation)
@progress 'Compiling C++ source code...' 71%
@progress 'Compiling C++ source code...' 86%
ERROR: UBT ERROR: Failed to produce item: /home/parnia/AirSim/Unreal/Environments/LandscapeMountains/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so
`
My clang version is 5.0.0-3 and unreal engine version is 4.18.
What should I do? Do I need to compile the whole AirSim before loading this new environment?

closed 2019-01-22 by msb336