microsoft/AirSim #758: Using Python and C++ APIS

olivia-skydio, 2018-01-26:
Hi there, 

I want to use the C++ API's to generate image data. I've looked at all the tutorials, but I'm a bit confused what directory I actually use to put my C++ (or python) files in that will get the images using the API. Would it be in our project directory? Or the Airsim directory? 

If anyone could provide examples or help it would be appreciated!

sytelus, 2018-01-26:
Generally the way I would do is:

1. Copy sample project from repo to your own folder. For example, copy [HelloDrone](https://github.com/Microsoft/AirSim/tree/master/HelloDrone) to C:\MyProjects.
2. In .vcproj file, change path for [AdditionalIncludeDirectories](https://github.com/Microsoft/AirSim/blob/master/HelloDrone/HelloDrone.vcxproj#L113) and [AdditionalLibraryDirectories](https://github.com/Microsoft/AirSim/blob/master/HelloDrone/HelloDrone.vcxproj#L139) to repo directory everywhere.
3. Run Unreal project (which is server). Compile and run above project (which is client).

olivia-skydio, 2018-01-27:
Since I'm using Linux I wouldn't be using Visual Studio, so I don't have a .vproj file. Any advice for linux use that doesn't require VS?

sytelus, 2018-01-30:
You can either use cmake ([see this example](https://github.com/Microsoft/AirSim/blob/master/cmake/HelloDrone/CMakeLists.txt)) or just command line switches to compiler to add include libraries and libs you want to link.

Domi47, 2018-02-07:
I have the same problem. Do you know already how to do it?

olivia-skydio, 2018-02-07:
I ended up using the python api since I didn't have to worry about compiling issues.

Domi47, 2018-02-08:
Thank u @olivia-skydio. Related to your first question "what directory I actually use to put my C++ (or python) files in that will get the images using the API. Would it be in our project directory? Or the Airsim directory?...if I do create another c++ file that uses the API must it be in the AirSim folder?" How did you solve that? That's the main problem i have now since I'm new with all this. I'm also using Linux. If you give me an example it would be appreciated.

olivia-skydio, 2018-02-08:
@Domi47  I am not sure about if you want to use the C++ api because I didn't end up using it, although I don't believe you should have to be in the AirSim Directory. I'm not totally sure because I used the python api and just had "import AirSimClient import *" at the top of my file and it worked fine even when it was placed in a completely separate directory. 

Sorry I can't be more helpful!

sytelus, 2018-02-13:
To use C++ API, you can put your client code anywhere. Just make sure to add path for include and libs to AirSim folder. Here's a simple example of how [HelloCar application uses cmake](https://github.com/Microsoft/AirSim/blob/master/cmake/HelloCar/CMakeLists.txt) to add include and libs folder. If you are not familiar with what above means then I would suggest not to use C++.

closed 2018-02-13 by sytelus

phildue, 2018-03-16:
I followed the instruction manual for linux. After compiling UnrealEngine and AirSim, running the HelloCar example on linux gives me
`/usr/bin/ld: cannot find -lAirLib
/usr/bin/ld: cannot find -lrpc
collect2: error: ld returned 1 exit status`
Can you maybe provide a full example that is linking properly? Or is something wrong on my system configuration?
Thank you already

phildue, 2018-03-16:
Solved it by adding
`link_directories(${AIRSIM_ROOT}/build_debug/output/lib/)`
before creating the executable and using the cmake version provided in the repository

vinbo, 2018-06-25:
is there any way in github to mark a certain issue for myself?