microsoft/AirSim #760: Error missing .dll when launching the project

Haotian-Zhang, 2018-01-26:
Hi All,
I'm facing the same problem with setting up the AirSim Plugins for my project. After I build the AirSim pulled from git successfully. I copied the Plugins folder to my project and edit the .uproject. When I opened the .uproject and it pops out the warning:
[image]

My compile results displayed like this:
[image]

Can someone please tell me how to solve this issue? Thank you!

Platform: Windows 10 x64 + VS 2015

sytelus, 2018-01-27:
That's normal message. Just click Yes and keep going.

haotiz, 2018-01-28:
I used to build the project on Linux, so it's a little bit different in Windows here. Thank you so much! @sytelus

closed 2018-01-30 by sytelus