microsoft/AirSim #763: Collisions not triggering form certain objects.

zamlz, 2018-01-28:
Hi, in the neighborhood environment, it seems that the bushes and and branches of trees don't register as collisions for the quadcopter. How would go about fixing this issue?

sytelus, 2018-01-30:
I think some foliage probably doesn't have collision boxes?

zamlz, 2018-02-02:
Are there plans to fix it on the next release, or is there some way I can fix it now?

sytelus, 2018-02-13:
This environment has proprietary assets so we can't distribute its source code. But you can purchase it from [Unreal Marketplace](https://www.unrealengine.com/marketplace/modular-neighborhood-pack) for $70 and experiment with collision boxes around above meshes. If you go that route, please let us know what you find!

jonyMarino, 2021-07-06:
In this way, you can collide with branches and leafs:
https://github.com/microsoft/AirSim/issues/3855#issuecomment-874984693

closed 2021-07-06 by jonyMarino