microsoft/AirSim #765: settings.json not changing anything

olivia-skydio, 2018-01-29:
I'm experiencing the scene image view to be very dark so I've been trying to change the brightness in capture settings through the settings.json file in my ~/Documents/Airsim folder. Whenever I change the settings I see no change in the scene even though when I launch a project it says loading camera settings from ~/Documents/AirSim/settings.json. I've tried to rebuild Airsim and resync the plugins folder to the projects but no luck. I've also tried using extreme values on the brightness settings and there's still no change. 

What should I be doing to actually make use of the settings? Am I missing something? Thanks!

olivia-skydio, 2018-01-29:
****

closed 2018-01-29 by olivia-skydio

simba155, 2018-03-31:
Hey, I met the same problem as you.
How did you do it???