microsoft/AirSim #766: AirSim and Intel Aero RTF

dimitrimav, 2018-01-29:
Hi, 
I am trying to connect AirSim to Intel Aero RTF through UDP and wifi. Using the Intel mavlink router [https://github.com/intel/mavlink-router]), mavlink packets are routed to 10.42.0.173:14550 (my computer’s ip is 10.42.0.173) and thus QGC can successfully connect to the flight controller. 
Then to connect AirSim to the vehicle instead of QGC, I have the following settings.json file:
{
  "SettingsVersion": 1.0,
  "DefaultVehicleConfig": "PX4",

  "PX4":{
    "UdpIp": "10.42.0.173",
    "UdpPort":14550,
    "UseSerial": false
  }
  }
However I get the error:
 [image]
I’m making sure that I have not opened QGC first so the port is not open.  I have also tried "UdpIp":10.42.0.1 (the Aero IP) but that does not work either.  Any ideas?

sytelus, 2018-01-30:
Are you able to get it working without QGC in the loop?

dimitrimav, 2018-01-30:
No, it does not work without QGC either. I also routed mavlink packets to another port to make sure that it's not taken by QGC (even if it's closed) but that does not work either.

sytelus, 2018-02-13:
Unfortunately we haven't worked on Intel Aero. Please do experiment and feel to contribute support for Intel Aero.

closed 2018-02-13 by sytelus

ulrikborjesson, 2018-02-21:
@dimitrimav I've been thinking of getting the intel rtf drone for an airsim project I'm doing since I don't wanna spend to much time building a custom drone, so I'm wondering if you got this to work?

dimitrimav, 2018-02-22:
@ulrikborjesson I believe what I was missing is that I had to set the LocalHostIp in settings.json to my computer's IP. However, I have not tried it. 

Keep in mind that you will have to change the PX4 airframe to HIL quadcopter via QGC every time you use the drone with Airsim, and then back to regular airframe when you want to fly it (meaning you have to redo the sensor calibrations).