microsoft/AirSim #768: MacOS master gives: Error at startup: mkdir failed for path  with errorno 2 and message No such file or directory: unspecified iostream_category error

asimonov, 2018-01-30:
I see error message 

"Error at startup: mkdir failed for path  with errorno 2 and message No such file or directory: unspecified iostream_category error"

when I start AirSim (master) in UE4Editor 4.17.2 (pulled using Epic Games launcher on Mac).
I have ~/Documents/AirSim/settings.json, but it does not help.

https://github.com/Microsoft/AirSim/issues/735 does not resolve my issues on Mac.
BTW, on another, Linux Ubuntu 16.04, box I can build and run master just fine.

asimonov, 2018-01-31:
potentially related to this commit

https://github.com/Microsoft/AirSim/commit/4fa3db0541acbe28ad45f67ca8412e37edfe435a

richriley, 2018-01-31:
I'm seeing the same issue.

asimonov, 2018-01-31:
I manage to work around the issue by replacing the following function in AirSim/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp (taken from last working versions):

```
void ASimHUD::initializeSettings()
{
    //TODO: should this be done somewhere else?
    //load settings file if found
    typedef msr::airlib::Settings Settings;
    try {
        FString settings_filename = FString(Settings::getUserDirectoryFullPath("settings.json").c_str());
        FString json_fstring;
        bool load_success = false;
        bool file_found = FPaths::FileExists(settings_filename);
        if (file_found) {
            bool read_sucess = FFileHelper::LoadFileToString(json_fstring, *settings_filename);
            if (read_sucess) {
                Settings& settings = Settings::loadJSonString(TCHAR_TO_UTF8(*json_fstring));
                if (settings.isLoadSuccess()) {
                    UAirBlueprintLib::setLogMessagesHidden(!settings.getBool("LogMessagesVisible", true));
                    UAirBlueprintLib::LogMessageString("Loaded settings from ", TCHAR_TO_UTF8(*settings_filename), LogDebugLevel::Informational);
                    load_success = true;
                }
                else
                    UAirBlueprintLib::LogMessageString("Possibly invalid json string in ", TCHAR_TO_UTF8(*settings_filename), LogDebugLevel::Failure);
            }
            else
                UAirBlueprintLib::LogMessageString("Cannot read settings from ", TCHAR_TO_UTF8(*settings_filename), LogDebugLevel::Failure);
        }

        if (!load_success) {
            //create default settings
            Settings& settings = Settings::loadJSonString("{}");
            //write some settings in new file otherwise the string "null" is written if all settigs are empty
            settings.setString("SeeDocsAt", "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md");
            settings.setDouble("SettingsVersion", 1.0);

            if (!file_found) {
                std::string json_content;
                //TODO: there is a crash in Linux due to settings.saveJSonString(). Remove this workaround after we only support Unreal 4.17
                //https://answers.unrealengine.com/questions/664905/unreal-crashes-on-two-lines-of-extremely-simple-st.html
#ifdef _WIN32
                json_content = settings.saveJSonString();
#else
                json_content = "{ \"SettingsVersion\": 1, \"SeeDocsAt\": \"https://github.com/Microsoft/AirSim/blob/master/docs/settings.md\"}";
#endif
                json_fstring = FString(json_content.c_str());
                FFileHelper::SaveStringToFile(json_fstring, *settings_filename);
                UAirBlueprintLib::LogMessageString("Created settings file at ", TCHAR_TO_UTF8(*settings_filename), LogDebugLevel::Informational);
            }
        }
    }
    catch (std::exception& ex) {
        UAirBlueprintLib::LogMessage(FString("Error loading settings from ~/Documents/AirSim/settings.json"), FString(ex.what()), LogDebugLevel::Failure, 30);
    }
}

```

asimonov, 2018-01-31:
but then, after successful build of Blocks in UE, I see SEGV crash trying to Play it:

```
LoginId:ff30f0dace4101ed5330f08c0e66c5f8
EpicAccountId:12c0fbb6ec7446d2ae4d9553142fe096

 SEGV_MAPERR at 0x0

ASimHUD::initializeSubWindows() Address = 0x18a7a597b [/Users/alexeysimonov/dev/px4/AirSim/Unreal/Environments/Blocks 4.17 - 2/Plugins/AirSim/Source/SimHUD/SimHUD.cpp, line 311] [in UE4Editor-AirSim.dylib]
ASimHUD::createMainWidget() Address = 0x18a7a46f1 [/Users/alexeysimonov/dev/px4/AirSim/Unreal/Environments/Blocks 4.17 - 2/Plugins/AirSim/Source/SimHUD/SimHUD.cpp, line 179] [in UE4Editor-AirSim.dylib]
ASimHUD::BeginPlay() Address = 0x18a7a2d23 [/Users/alexeysimonov/dev/px4/AirSim/Unreal/Environments/Blocks 4.17 - 2/Plugins/AirSim/Source/SimHUD/SimHUD.cpp, line 28] [in UE4Editor-AirSim.dylib]
AActor::DispatchBeginPlay() Address = 0x10c3eb988 (filename not found) [in UE4Editor-Engine.dylib]
AWorldSettings::NotifyBeginPlay() Address = 0x10de20ac4 (filename not found) [in UE4Editor-Engine.dylib]
AGameStateBase::HandleBeginPlay() Address = 0x10ce4fff1 (filename not found) [in UE4Editor-Engine.dylib]
UWorld::BeginPlay() Address = 0x10de006ee (filename not found) [in UE4Editor-Engine.dylib]
UGameInstance::StartPlayInEditorGameInstance(ULocalPlayer*, FGameInstancePIEParameters const&) Address = 0x10ce22b3d (filename not found) [in UE4Editor-Engine.dylib]
UEditorEngine::CreatePIEGameInstance(int, bool, bool, bool, bool, float) Address = 0x1135d3d3c (filename not found) [in UE4Editor-UnrealEd.dylib]
UEditorEngine::PlayInEditor(UWorld*, bool) Address = 0x1135c948b (filename not found) [in UE4Editor-UnrealEd.dylib]
UEditorEngine::StartQueuedPlayMapRequest() Address = 0x1135b1857 (filename not found) [in UE4Editor-UnrealEd.dylib]
UEditorEngine::Tick(float, bool) Address = 0x112d3fe64 (filename not found) [in UE4Editor-UnrealEd.dylib]
UUnrealEdEngine::Tick(float, bool) Address = 0x113a2228c (filename not found) [in UE4Editor-UnrealEd.dylib]
FEngineLoop::Tick() Address = 0x10a12de75 (filename not found) [in UE4Editor]
GuardedMain(wchar_t const*) Address = 0x10a135c22 (filename not found) [in UE4Editor]
-[UE4AppDelegate runGameThread:] Address = 0x10a1435d0 (filename not found) [in UE4Editor]
-[FCocoaGameThread main] Address = 0x10a409446 (filename not found) [in UE4Editor-Core.dylib]
Unknown() Address = 0x7fff904d9f24 (filename not found) [in Foundation]
_pthread_body Address = 0x7fff922e999d (filename not found) [in libsystem_pthread.dylib]
_pthread_body Address = 0x7fff922e991a (filename not found) [in libsystem_pthread.dylib]
thread_start Address = 0x7fff922e7351 (filename not found) [in libsystem_pthread.dylib]
```

asimonov, 2018-01-31:
so may be more changes need reverting for Mac to work again

asimonov, 2018-02-01:
looks like https://github.com/Microsoft/AirSim/pull/774 may fix the 'mkdir' problem on Mac

sytelus, 2018-02-13:
https://github.com/Microsoft/AirSim/pull/774 is now merged with master. Please reopen is you see same issue again.

closed 2018-02-13 by sytelus

KazW, 2018-03-27:
I get this issue as well when executing a packaged build on a machine other than the one that built it. I'm trying to replicate Chris Lovett's YouTube tutorial with the Landscape Mountains environment, but on a Mac.

Looking at the error/code above, I thought creating a `~/Documents/AirSim/settings.json` might prevent it, but it didn't.

I'm using:
UE: 4.17.2
Xcode: 8.3.2
OS: 10.13.3 (for both builder and client)
AirSim: master
Build Type: Debug/Development/Shipping (all have the same result)

[Shipping build and settings.json](https://www.dropbox.com/sh/gmg9q2euyz65h4f/AACVGIg_Ewv2dk7QZ6BmFJ2Ba?dl=0)

Below are is error that happens shortly after startup and where the game hangs after clicking "ok" on the dialog.
[image]
[image]