microsoft/AirSim #772: DrivetrainType.ForwardOnly mode is not working properly as expected

akshaysw, 2018-01-31:
@sytelusI did try command `client.moveByVelocity(-5, -5, -5, duration=20, drivetrain = DrivetrainType.ForwardOnly, yaw_mode=YawMode(False, 0))` but its not working properly as explained in your [docs](https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#drivetrain) 

> drivetrain
There are two modes you can fly vehicle: Drivetrain = ForwardOnly and Drivetrain = MaxDegreeOfFreedom. When you specify ForwardOnly, you are saying that vehicle's front should always point in the direction of travel. So if you want drone to take left turn then it would first rotate so front points to left. This mode is useful when you have only front camera and you are operating vehicle using FPV view. This is more or less like travelling in car where you always have front view. The MaxDegreeOfFreedom means you don't care where the front points to. So when you take left turn, you just start going left like crab. Quadrotors can go in any direction regardless of where front points to. The MaxDegreeOfFreedom enables this mode.

> yaw_mode
yaw_mode has two fields, yaw_or_rate and is_rate. If is_rate = true then yaw_or_rate is interpreted as angular velocity in degrees/sec which means you want vehicle to rotate continuously around its axis at that angular velocity while moving. If is_rate = false then yaw_or_rate is interpreted as angle in degrees which means you want vehicle to rotate to specific angle (i.e. yaw) and keep that angle while moving.
You can probably see that YawMode::is_rate = true is pointless in ForwardOnly mode because you are contradicting by saying that keep front pointing ahead but also rotate continuously. However if you have YawMode::is_rate = false in ForwardOnly then you can do some funky stuff. For example, you can have drone do circles and have YawMode::yaw_or_rate = 90 so camera is always pointed to center ("super cool selfie mode"). In MaxDegreeofFreedom, you can get some funky stuff by setting YawMode::is_rate = true and say YawMode::yaw_or_rate = 20. This will cause drone to go in its path while rotating which may allow to do 360 scanning.
In most cases, you just don't want yaw to change which you can do by setting yaw rate of 0. The shothand for this is YawMode::Zero().

i think for every i second of total duration quadcopter rotates by angle 90 degrees clockwise or anticlock wise depends on direction where you are moving quadcopter i will try to explain below with screenshots of code and drone pictures(arrow in the picture shows front direction of drone(front camera)
this is my start position

[image]

now i will try to move in -ve Y-dirction for 3 seconds according to your doc its camera should face -ve Y drection but its facing +ve Y direction `client.moveByVelocity(0, -5, 0, duration=3, drivetrain = DrivetrainType.ForwardOnly, yaw_mode=YawMode(False, 0))`

[image]

now if i again try to move in -ve Y-direction for one second camera of drone faces +ve X-direction `client.moveByVelocity(0, -5, 0, duration=1, drivetrain = DrivetrainType.ForwardOnly, yaw_mode=YawMode(False, 0))`

[image]

now i try to move in +ve Y-direction for two seconds it faces -ve X-direction `client.moveByVelocity(0, 5, 0, duration=2, drivetrain = DrivetrainType.ForwardOnly, yaw_mode=YawMode(False, 0))`

[image]

isnt front camera of drone should face dirction of travel in DrivetrainType.ForwarOnly mode according to your doc or is there something i'm doing wrong?

sytelus, 2018-02-13:
One possibility is that you are running command for very short durations. In forward only mode, drone needs to rotate and we can't do that instantaneously. This is because drones don't have breaking friction so if you rotate too fast then it would oscillate. So you need to give some time to stabilize, especially if you are doing about turns!

closed 2018-04-12 by sytelus

sytelus, 2018-04-12:
FYI - We had a bug that caused yawing unnaturally for the drone. It's fixed now.