microsoft/AirSim #776: Segmentation API is limited

aburgm, 2018-01-31:
The `simSetSegmentationObjectID` API as described [here](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#segmentation) is somewhat limited, in particular:

 * It is not possible to un-assign a segmentation ID so that it does not show up in the segmentation view (custom depth rendering disabled). In the neighbourhood environment, there are some leaves lying on the roads or the grass which I would like to just ignore for the segmentation so that they take on the segmentation ID of the underlying road or grass, respectively.

 * There are objects that do not have a reasonable (or easily to look-up) object ID, such as spline mesh components or mesh components instantiated by a foliage actor. Especially spline mesh components have the same name as the landscape, so there is no way to distinguish them from the landscape. This applies to the road in the mountains landscape environment, or the power lines, fences or the path in the park in the neighbourhood environment.

aburgm, 2018-01-31:
For the first point, an easy extension could be to turn off the custom depth rendering flag when the object ID passed to the API is negative.

For the second point, I often find myself in the situation where I want all instances of a particular mesh, such as "Path_01" or "Tree_01", to have the same object ID. I have played around a bit with changing the [UAirBlueprintLib::GetMeshName](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/AirBlueprintLib.cpp#L182) function to the following:

```c++
template<class T>
std::string UAirBlueprintLib::GetMeshName(T* mesh)
{
    UStaticMeshComponent* staticMesh = dynamic_cast<UStaticMeshComponent*>(mesh);
    if (mesh->GetOwner() && staticMesh && staticMesh->GetStaticMesh())
        return std::string(TCHAR_TO_UTF8(*(mesh->GetOwner()->GetName()))) + "/" + TCHAR_TO_UTF8(*staticMesh->GetStaticMesh()->GetName());
    else if (mesh->GetOwner())
        return std::string(TCHAR_TO_UTF8(*(mesh->GetOwner()->GetName()))) + "/null";
    else
        return ""; // std::string(TCHAR_TO_UTF8(*(UKismetSystemLibrary::GetDisplayName(mesh))));
}
```

This plays together nicely with the regex functionality of the `simSetSegmentationObjectID` API because it easily allows to match either the owner name or the static mesh name or both. Obviously this would break existing users of this API, so this is meant more as a starting point for discussion. Thoughts welcome!

sytelus, 2018-02-13:
I agree that we should have API to unset segmentation object ID. May be we should -1 as signal to unset. Please feel free to contribute this change :).

For the leaves on road, I think its incorrect to change segmentation. If you don't want leaves then you might want to remove it from the environment. Segmentation should always shows the ground truth and not the "convenient groundtruth".

 For the change above, I think what is happening is that you only end up using mesh owner's name because algorithm considered only first few chars of name to generate hash (to avoid usual training numbers). It was bad algorithm and I've now replaced it with one that eliminates numericals and some punctuation. So if you try latest version, you will see each owner might end up with different colors for each mesh. Fortunately, I've now added settings to control segmentation behaviour. So may be we should add settings that says "use only owner name for hash" vs "mesh name". That way you can switch between two behaviours. Please also feel free to contribute this change if you think its important. BTW, you don't want to dynamic_cast because its expensive and this function gets called in loop.

aburgm, 2018-02-14:
Actually I'm not so worried about the automatically assigned object IDs using the hash, but what I can match using the simSetSegmentationObjectID API. For example, without the static mesh name, I cannot reliably match the road pieces in the LandscapeMountains environment to assign them a certain ID.

Note also that there are three different names: The owner name, the mesh name, and (for static mesh components), the static mesh name. The static mesh name is the name of the mesh as it appears in the content browser in the editor. 

For the road in the LandscapeMountains environment, these happen to be: `Landscape_10` (owner name), `SplineMeshComponent_<n>` (mesh name), `SM_Path` (static mesh name). The owner name is the same as for the landscape itself, so it cannot be used to give the landscape a different ID than the road. The mesh names are all distinct by a unique number, so could be used, but they are not very intuitive, especially if there is more than one spline with different meshes in the landscape (e.g. in the neighbourhood environment). The static mesh name is typically fairly intuitive, but the same for all instances of the same static mesh. So for instance, two instances of the same tree could not be assigned different IDs.

I like the idea of specifying which name to use in the segmentation settings.

hrdisaac, 2018-05-18:
The stencil buffer has just 8 bits and therefore it only supports up to 256 segmentation IDs. I want to have more than 256 tags for my environment. Is there any way to fix this?

sytelus, 2018-10-17:
The new segmentation APIs now allows to unset the object ID. Also identifying objects is now have better documentation. Please file a new issue if there is still a need for further changes.

closed 2018-10-17 by sytelus

tchaton, 2019-04-24:
Hello,

Is it possible to get all moving objects ( cars, pedestrians ) with a unique label ?

Best,
T.C