microsoft/AirSim #778: Q matrix for pointcloud - Depth code implementation

badabook, 2018-01-31:
1_ I work in Python (Car Mode), and I wanted to reconstruct the pointcloud from the Depth map, so i need the Q matrix, which is:
```
Q = np.array([
        [1, 0, 0, -Width/2],
        [0, 1, 0, -Height/2],
        [0, 0, 0, focal_length],
        [0, 0, -1/B, 0]
    ])
```
I know that:
FOV = 90
focal_length = Width/ 2 / tan(FOV/2) = Width/2
B is the baseline.

Now, I have carefully examined every file in AirsimPlugin, and I found this:

```
// Create In-Car camera component 
    InternalCameraBase1 = CreateDefaultSubobject<USceneComponent>(TEXT("InternalCameraBase1"));
    InternalCameraBase1->SetRelativeLocation(FVector(36.0f, 0, 50.0f)); //center
    InternalCameraBase1->SetupAttachment(GetMesh());
    InternalCameraBase2 = CreateDefaultSubobject<USceneComponent>(TEXT("InternalCameraBase2"));
    InternalCameraBase2->SetRelativeLocation(FVector(36.0f, -10, 50.0f)); //LEFT
    InternalCameraBase2->SetupAttachment(GetMesh());
    InternalCameraBase3 = CreateDefaultSubobject<USceneComponent>(TEXT("InternalCameraBase3"));
    InternalCameraBase3->SetRelativeLocation(FVector(36.0f, 10, 50.0f)); //RIGHT
    InternalCameraBase3->SetupAttachment(GetMesh());
    InternalCameraBase4 = CreateDefaultSubobject<USceneComponent>(TEXT("InternalCameraBase4"));
    InternalCameraBase4->SetRelativeLocation(FVector(25, -10, 75.0f)); //driver
    InternalCameraBase4->SetupAttachment(GetMesh());
    InternalCameraBase5 = CreateDefaultSubobject<USceneComponent>(TEXT("InternalCameraBase5"));
    InternalCameraBase5->SetRelativeLocation(FVector(-36.0f, 0, 50.0f)); //rear
    InternalCameraBase5->SetRelativeRotation(FRotator(0, 180, 0));
    InternalCameraBase5->SetupAttachment(GetMesh());
```

So I suppose that LEFT and RIGHT are the two cameras that I can use to capure Stereo Images, right?
From this data I think that the baseline should be 10+10 = 20.  Am I wrong?
Obviously I read the example [point_cloud.py](https://github.com/Microsoft/AirSim/blob/master/PythonClient/point_cloud.py), but I think this matrix is wrong, because with B = 20 and the Q matrix above I obtained great results.

Than, if I am right, I would like to get this parameters via some funcion, instead of going reading the code.. is there a way to get camera parameters?

2_ I use DepthPlanner mode to generate Depth map, but I would like to know which is the difference between every ImageType, because I found not clear the explaination given in the documentation (I did't say that it is wrong, but is difficult for me to understand it from there, and I would like a more detailed explaination).

3_ My most important question is: where can I find HOW the DepthPlanner, DepthPerspective, DisparityNormalized,..  are created??? 
I mean the implementation, because when I call this api
`client.simGetImages([ImageRequest(0, AirSimImageType.DepthPlanner, True, False)])`
I can't find the process which generates a DepthPlanner image, so is this code accessible??
I know that is an Enum, but there must be a function that implements this somewhere.

PS: I tryed reading these issues:
[From AirSim](https://github.com/Microsoft/AirSim/issues/269)
 [From UnrealCV](https://github.com/unrealcv/unrealcv/issues/14)
But I did't find anything useful

I really need to solve these problems, and I think this will be usefull for a ot of people.
Thanks in advice

sytelus, 2018-02-13:
You are correct, the baseline is 20cm. The point_cloud.py was contributed by another user and unfortunately we didn't had throughly to test it but thought others might find it as good starting point.

I agree that we should have APIs to get camera parameters. I'll add this in list of features.

DepthPlanner basically projects vector to the point in the world on a perpendicular to the camera plan. Here's Unreal material code to generate it:

[image]

DepthProjection gives you the length of vector from camera to the point in the world.

[image]

Both depth views are generated using Unreal materials which gets compiled as shaders. Fortunately Unreal has visual tool to work with this so not much expertise needed. You can find these shaders [here](https://github.com/Microsoft/AirSim/tree/master/Unreal/Plugins/AirSim/Content/HUDAssets). To view them, use ["View Plugin Content"](working_with_plugin_contents.md) and then navigate to Content/HUDAssets folder. You will also see material functions where you can find above code.

sytelus, 2018-02-21:
FYI – I just added ability to get camera parameters via APIs in AirSim (1st bullet point below). Here’s [more info in our docs](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#getting-camera-parameters).

shmm91, 2018-12-12:
@badabook Have you generated point cloud successfully? 
Can you help me, I am confused with the issues about point cloud.
Image width:640, Image height:480, FOV:90
This is my python program.
[pcl.txt](https://github.com/Microsoft/AirSim/files/2671327/pcl.txt)

The point cloud result display in RVIZ
[image]

shmm91, 2018-12-12:
Sorry for my fault, this is the raw image.
[image]

chaitanyantr, 2019-02-12:
@shmm91 

Can you please try to generate the point cloud by not viewing the sky .

go to multi-rotor mode and down the gimbal to 35 degree and then try to generate the point cloud with depth...and plot it on Rviz.

dnychennnn, 2020-06-25:
Hi, is there a standard way to generate the point cloud from disparity normalized given by AirSim? And how should the Q matrix be changed while the multi-rotor is flying, or it is not need to?

chaitanyantr, 2020-06-26:
i have used projection matrix and depth from 'depth float' data from air sim to generate point cloud.
I suppose there is a script in airsim pkg folder named point cloud.

chaitanyantr, 2020-06-26:
projectionMatrix = np.array([[-0.501202762, 0.000000000, 0.000000000, 0.000000000],
                              [0.000000000, -0.501202762, 0.000000000, 0.000000000],
                              [0.000000000, 0.000000000, 10.00000000, 100.00000000],
                              [0.000000000, 0.000000000, -10.0000000, 0.000000000]])

chaitanyantr, 2020-06-26:
https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/point_cloud.py

dnychennnn, 2020-06-26:
@chaitanyantr I know, but shouldn't the reprojectImageTo3D() function use the disparity map for reconstruction(not sure if it need to be de-normalized)? When I try depthplanner, depthperspective, disparity with the Q matrix I got from the formulas, I only received reasonable results with depth planner( depth perspective being too curved, disparity being the contour of the objects ), which doesn't follow the theory behind this function at all. How's the standard way to proceed with this point cloud reconstruction problem?