microsoft/AirSim #780: Unable to debug - DebugGame Editor error

jjlee1991, 2018-02-01:
Followed the step by step, till pressing for "F5" to debug, when I faced the error:
"The following modules are missing or built with a different engien version:
UE4Editor-LandscapeMountains-Win64-DebugGame.dll
Would you like to rebuild them now?"

Selected yes, and this error occurred:
"LandscapeMountains could not be compiled. Try rebuilding from source Manually".

Please advice
Thank you

sytelus, 2018-04-12:
Can't reproduce. I'm wondering if your environment is not setup properly.

closed 2018-04-12 by sytelus

samby707, 2018-12-20:
I am also facing the same issue. please help

changh95, 2019-11-26:
[image]
Facing the same issue here.