microsoft/AirSim #783: Problem to run <build.cmd> about CMake

Kim-BongSu, 2018-02-02:
When I followed the process of build AirSim, There were some problems. 
put the text build.cmd on command line, I got the message like this.
I'm not familiar in Visual Studio & Programming like this please help me.
##_-------------------------------------------------------------------------------

C:\Users\kbs\source\AirSim>build.cmd
Found cmake version: 3.10.2
Starting cmake to build rpclib...
-- Selecting Windows SDK version  to target Windows 10.0.16299.
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "C:/Users/kbs/source/AirSim/external/rpclib/rpclib-2.2.1/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/kbs/source/AirSim/external/rpclib/rpclib-2.2.1/build/CMakeFiles/CMakeError.log".
Microsoft (R) Build Engine Version 14.0.25420.1]
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: There are no project files.
Swtich: ALL_BUILD.vcxproj
Microsoft (R) Build Engine Version 14.0.25420.1]
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: There are no project files.
Switch: ALL_BUILD.vcxproj

#### Build failed - see messages above.

##_-------------------------------------------------------------------------------

sytelus, 2018-02-13:
My guess is that your Visual Studio 2015 Update 3 is not installed correctly OR you haven't installed Visual C++. Also you need to run this from VS2016 Native x64 command prompt. Please search for Visual studio documentation for further info.

closed 2018-02-13 by sytelus

gitstar12, 2018-02-16:
I am having the same problem. I am using VS2015 installed c++

sytelus, 2018-02-17:
You don't seem to have C++ compiler. See the error messages you are getting:

```
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
```

Also you need to run this from VS2016 Native x64 command prompt.

ElCodeMonkey, 2019-03-20:
I get the same problem and I'm using the Native x64 command prompt. Instructions say to install VS 2017, not 2015. Yet if I open Cmake trying to see if there was a way to reference my c/c++ compiler, it seems to have a dropdown for VS versions and none of them are 2017. So... are the instructions just wrong?