microsoft/AirSim #784: Propellers render outside MultiRotor

tachitachi, 2018-02-02:
I've noticed that every time I run DQNdrone.py in the released LandscapeMountains environment, if I leave it running long enough, the propellers end up outside of the MultiRotor.
[image]
You can see the red and white propellers off to the sides of the MultiRotor. They even show up in the camera views, which would be problematic for training.

This was using the v1.1.7 release of LandscapeMountains in Windows 10. I'm not sure if this is a global issue, or an issue with that particular environment, or if it's some setting on my end.

sytelus, 2018-02-13:
This is very strange and have never seen this happening myself. I'm assuming you are using unmodified binaries. Any idea after how long this happens? Is this very consistent?

tachitachi, 2018-02-13:
Yes, I'm using unmodified binaries. I'm not sure exactly how long it takes before this happens, but it does happen every time after letting it run overnight. The longer it runs, the further the propellers spread out. It doesn't seem to happen if I call the reset function regularly. But DQNdrone.py doesn't use the reset function.

droghio, 2020-04-30:
I can confirm I have seen this issue using AirSim v1.2.0 (pulled from master in January 2020) and UE4.20. Exact placement of the rotor varies but they move outside of the frame. When I have run overnight I have seen this issue consistently, but I do not have an exact time to failure. It seems to be on the order of hours. This has reproduced on both the Blocks and on custom environments.

[image]

agentfeux, 2020-09-14:
I have this bug too. To work around the issue, for the training I use the option in unreal to hide the pawn from rendering so it hides the drone but still moves normally in the scene. Hope it helps some people.