microsoft/AirSim #787: AirSim needs to simulate accelerometer bump on landing

lovettchris, 2018-02-03:
PX4 (SITL or HITL) doesn't detect "landing state" because AirSim doesn't simulate the bump in acceleration that should happen when the drone hits the ground.

cdyk, 2018-02-14:
An idea: One approach might be to compute the acceleration in the collision detection (from the timestep size and change in velocities determined by the collision impulse). Since the collision detection runs after time-integration, the current acceleration isn't used anymore as far as I can see. 

Then ImuSimple could pick this up.

Directly differentiation the velocities from impulses will create spikes of acceleration with magnitude scaled by the step size for each collision event. Though, on average I think it should be pretty correct, so the results must be averaged somehow (maybe in ImuSimple). I guess real IMUs have impulse responses that may hint at an appropriate filter window size?

cdyk, 2018-02-15:
I noticed that FastPhysicsEngine sets the linear acelleration to zero and not gravity when grounded, which in turn makes the IMU report an upwards acceleration of 1g when grounded.

Without ground_lock enabled, I get a lot of pulses of upwards acceleration (from the collision responses) in between resting which really messes with the calibration of the drone.

Fixing this made the PX fail to differentiate between free fall and landed, so I assume using zero was deliberate used?

sytelus, 2018-04-12:
Just saw this was implemented. Should we close this now?

lovettchris, 2018-12-23:
Yes, works great now, thanks.

closed 2018-12-23 by lovettchris