microsoft/AirSim #790: Access to information about the City environment using the API only

miev88, 2018-02-05:
In general, it is possible to retrieve the location of other cars and pedestrians on the road, if you dig in to Unreal Engine programming #232. Moreover, it is possible to retrieve the object pose using the API and the Unreal Engine #728.

Now I have a specific question about the City environment. Is it possible to retrieve information about how many cars are on the road (number and location), which trajectories the are following, how many crossings there are in the environment and their location using the API only?

If it is not, is there a plan to add this feature?

PS: How is the trajectory for other cars on the road computed?

sytelus, 2018-02-13:
Unfortunately, this information is not available right now. We are considering several options to generate this information but nothing has solidified yet. One approach is to use OpenDrive format to export this information. Other approach is to generate 2D map as image. If you have any other ideas, please feel free to chip in.

AndreasKernAudi, 2018-11-15:
An OpenDRIVE Export of the roads, the signals and the surrounding objects would be great!

ironclownfish, 2019-09-27:
There are a couple new APIs that can help with this.
simListSceneObjects(regex) will give you a list of objects in the scene, so you can use a regex to find all stoplights, cars, etc. by name patterns.
simGetObjectPose(obj_name) can then be called with the object names found from simListSceneObjects().
So together, they provide a pretty straightforward way to discover sets of objects and ask for pose information about them.

closed 2019-09-27 by ironclownfish

nightduck, 2020-08-26:
simListSceneObjects only lists stationary objects. The moving cars and pedestrians in the CityEnviron environment aren't included. Presumably they're managed by the CityEnv_PedManager_26 and CityEnviron_NPCVehicleManager objects, but Airsim's API will only let you access stationary cars, and none of the pedestrians.