microsoft/AirSim #794: Build Error : MavLinkCom.vcxproj

Kim-BongSu, 2018-02-06:
c:\program files (x86)\microsoft visual studio 14.0\vc\include\crtdefs.h(10): fatal error C1083: Cannot open including files 'core
crt.h': No such file or directory [C:\Users\kbs\source\AirSim2\MavLinkCom\MavLinkCom.vcxproj]

sytelus, 2018-02-09:
I'm guessing your Visual Studio 2015 either doesn't have Update 3 or has only partial VC++ product installed.

closed 2018-02-09 by sytelus

Isaac45, 2018-05-07:
having the same problem, using visual studio 2017
c:\program files (x86)\microsoft visual studio 14.0\vc\include\crtdefs.h(10): fatal error C1083: Cannot open including files 'core
crt.h': No such file or directory [C:\Users\kbs\source\AirSim2\MavLinkCom\MavLinkCom.vcxproj]

saihv, 2018-05-10:
Confirmed, I have the same problem while trying to build with Visual Studio 2017.