microsoft/AirSim #796: Best way to stream scene and depth over the network

realdealneil, 2018-02-06:
I'm trying to find the best way to export images over the network (to test vision algorithms on an embedded computer).  I've written a way of sending the rendered output from AirSim using gstreamer and H264 compression from my windows PC to an Nvidia Jetson TK1.  This allows me to send frames at 20-25 fps (monochrome).  I do this using an existing gstreamer element that grabs a portion of the screen on your computer...I grab the upper right 1280x720 pixels and position my render window there.  

Now I'd like to be able to send both the FPV view and the depth truth (DepthPlanner probably) at the same time, and gstreamer doesn't seem like the ideal way to do this, so I started looking into the image API provided by AirSim.  I haven't dug all the way into how it works, but I can get images from one or more cameras over the Multirotor RPC client.  However, the fastest I've been able to get these images to come in is at 10 fps (on an Nvidia GTX1060).  This is without saving the images...just rendering and capturing and throwing them away.  Is there a way to do better than 10 fps?  I suppose I can slow the simulator down to achieve higher effective frame rates for post processing, but ideally I'd like to achieve 20+ fps.

zergler, 2018-02-07:
I am trying to do something similar. I am wondering if there is a way of turning off rendering of the game viewports completely and only render a camera's scene once a request has been sent via the Multirotor RPC client. I think this would dramatically increase FPS now that the game doesn't have to render the entire scene in real time, only scenes from individual cameras which I can set to low resolutions. Does anyone know how to do this?

nnayeer, 2018-02-12:
Seconded @zergler , I would like to know how to turn off rendering the scene ( a sort of "headless mode") and still get the cameras to send images through API calls. Right now it seems if Unreal is in the background and not rendering the scene, the images returned through API calls do not update even if the vehicle is moving.

sytelus, 2018-02-13:
Zergler's idea is worth trying!

**Possibility 1**
Set [ViewMode setting](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#viewmode) to "Fpv". This will cause one scene rendering from only one point of view. Hope is that Unreal Engine would optimize this... Failing that, see below.

**Possibility 2**
Disable main camera entirely. In [showToScreen](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/PIPCamera.cpp#L94) method, comment everything out and replace that code with just one line `disableMain();`. Basically this will make sure the main camera never ever gets activated and only subwindows have rendering.

sytelus, 2018-02-17:
I added this change to deactivate all cameras: https://github.com/Microsoft/AirSim/commit/e67a990798e3bf96fb503403c620083ec0a4dda0

Unfortunately it isn't working. It seems like Unreal just doesn't want all cameras to be disabled! So I've posted a question here: https://answers.unrealengine.com/questions/759610/how-to-completely-disable-any-camera.html

Please feel free to play around with above change. If you find anything interesting let me know.

zergler, 2018-02-23:
@sytelus Thanks for the tip! We are able to render 640x480 scene at around 60-130 FPS by turning the resolution factor all the way down, depending on settings. This makes the viewport highly blocky, but it doesn't affect screenshot resolution, as you've said. It would be nice to go even faster though.

sytelus, 2018-03-02:
Hi folks, @aburgm submitted a [great PR](https://github.com/Microsoft/AirSim/pull/850) that now allows to freeze main viewport rendering while still recording images! This is checked in to master so please give it a try. You just need to set ["ViewMode": "NoDisplay" in settings.json](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#viewmode).

realdealneil, 2018-03-02:
That's great!  I'll check it out

Neil

closed 2018-04-12 by sytelus

coldwarrior5, 2018-12-31:
Is it not possible to achieve higher fps by using shared memory similar to Deepdrive simulator?

> Our frame rates are higher while capturing several cameras, due to the use of shared memory rather than sockets and asynchronous transfer.
More information available at: https://deepdrive.io/blog/building-on-unreal.html
Source code available at: https://github.com/deepdrive/deepdrive