microsoft/AirSim #799: Drone flies erratically in Mission mode or using Hello_drone.py (PX4 HITL)

dimitrimav, 2018-02-07:
Hello, 

I am running Airsim with a Pixhawk flashed with PX4 1.5.5 in the loop. The flight is fairly stable in Manual flight modes (e.g. Manual/Stabilized) using a remote controller. However, it starts flying erratically in any form of autonomous or assisted mode I have tried. It just bounces all around and ends up crashing in the end. 
For example:
	- When trying to execute a waypoint mission using QGroundControl. The drone does respond in some way but hardly takes off the ground and when it does it crashes soon after.
	- When running Hello_drone.py. On commanding take-off the drone again just bounces. And of course autonomous motion does not work either ( client.moveToPosition(-10, 10, -10, 5)). 
	
Do you have any idea why this might be happening and how I can fix it?

sytelus, 2018-02-13:
For the time being we have paused developing for PX4 due to recent several issues. If you are experienced with PX4, you are welcome to help debug these issues. Few things I suspect,

1. New addition of distance sensor. This was contributed by someone else and wasn't originally supported by 1.1.5. So you might want to try disabling it. Here's [the line](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotorParams.hpp#L36) you want to change to false.

2. There was one issue we had noticed: accelerometer shows wild fluctuation while vehicle is on the ground. This might be throwing off PX4's calibration algos.

dimitrimav, 2018-02-23:
Thank you for the reply @sytelus.

Now I changed the physics rate from 300Hz to 500Hz in getPhysicsLoopPeriod() in SimModeWorldBase.cpp and it works perfectly! Both QGC missions and Hello_drone.py. 
I am now using the latest stable PX4 release (1.7.3), I don't know about 1.5.5.

closed 2018-02-23 by dimitrimav