microsoft/AirSim #800: simGetImages() synchronization issue for stereo images

olivia-skydio, 2018-02-07:
Referring to [this](https://github.com/Microsoft/AirSim/issues/244) previous issue, it seems like simGetImages() should be getting left/right /center /cameras during the same tick. Although when I'm getting these images I'm checking the positions of the distance between the left/right cameras(based on x,y,z coordinates), and sometime's it's up to a cm off (24 cm) versus the 25 cm it's suppose to be which can cause issues when trying to generate stereo data. When camera->getActorLocation() is [called](https://github.com/Microsoft/AirSim/blob/ccd10306d91b4b2129d02972f6fb5a44fb44794e/Unreal/Plugins/AirSim/Source/UnrealImageCapture.cpp#L73), how do we know it's called in the same tick and that the it's not getting called later when the camera might have moved? 

Is there a way to mediate this while still using simGetImages() vs the record button for simultaneous recording (personally prefer using the API since more flexibility). 

Thanks!

sytelus, 2018-02-13:
Do you mean camera_position from API response for two camera is not exactly 25cm always? That would be strange... Here's [the code that sets camera_position](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/UnrealImageCapture.cpp#L73). It would be great if you could help debug this. I've given some code explanation below.

We first call render_request.getScreenshot on [this line](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/UnrealImageCapture.cpp#L61). This call gets all the image data you requested for. Then we have loop that simply sets each camera position in to response. These positions are set in very quick succession so unless vehicle is moving very very fast you should get 25cm.

Are you using drones? Do you experience this when drone is moving fast? One thing I suspect is that physics engine keeps running on other thread so it might update position between two subsequent calls to the loop. If that's the case, the solution would be to pause physics engine before screenshot.

olivia-skydio, 2018-02-13:
@sytelus I realized this issue happens sometimes when there is a collision. You can continue to set the vehicle pose to a new position (in computer vision mode) but the camera poses do not match up to the vehicle (or sometimes each other). This is true even when ignoreCollision is set to true ( I believe it's the same bug as this [one](https://github.com/Microsoft/AirSim/issues/598) where the ignoreCollision doesn't work. I fixed this by manually double checking to make sure it never has a collision for any of the poses I set. 

I am using Computer Vision mode for the drone and do a loop of simSetPose() with simGetImage requests in each loop.

sytelus, 2018-02-15:
Thanks for the tip. I'll have to investigate this.