microsoft/AirSim #802: Packaging issue on ubuntu 16.04

gkbell46, 2018-02-08:
Hi I am new to Unreal engine and Airsim. I have followed the (link)[https://www.youtube.com/watch?v=OA4JUSwH98Q] for packaging. I get the error 

"UATHelper: Package (Linux): UnrealBuildTool: error: defination of macro 'PLATFORM_EXCEPTIONS_DISABLED' differs between the precompiled header ('1') and the command line ('0')"  

I am using 
Ubuntu 16.04
Unreal engine 4.17

Kindly requesting someone to help. TIA

sytelus, 2018-02-09:
Generally packaging from Linux is not recommanded. You probably want to use cross platform toolchain to make Linux build from within Visual Studio: https://docs.unrealengine.com/latest/INT/Platforms/Linux/GettingStarted/

For above error, you might want to post in Unreal forums: http://answers.unrealengine.com/

gkbell46, 2018-02-10:
@sytelus Thanks for your quick reply. Packaging works fine for a demo application created without AirSim plugin. I see this error only packaging along with AirSim plugin. I see similar error on Windows too.

kshitiz38, 2018-05-31:
I had the same problem in packaging for Linux environment...
A quick workaround is to disable exceptions in `Blocks.Build.cs` by setting `bEnableExceptions = false`, but remember this is not a fix just a workaround

TritonSailor, 2018-06-10:
@kshitiz38 That workaround didn't work for me. Disabling exceptions in Blocks.Build.cs didn't solve the above error.  I also tried disabling exceptions in AirSim.Build.cs. It generated many errors similar to:

In file included from /home/chris/Documents/Unreal Projects/Blocks 4.20/Plugins/AirSim/Source/AirLib/include/common/Settings.hpp:19:
/home/chris/Documents/Unreal Projects/Blocks 4.20/Plugins/AirSim/Source/AirLib/include/common/common_utils/FileSystem.hpp:173:13: error: 
      cannot use 'throw' with exceptions disabled
            throw std::ios_base::failure(std::strerror(errno));

TritonSailor, 2018-06-10:
This is definitely an AirSim plugin issue.  I've been able to package projects on linux without AirSim plugin.

I was able to get around the original issue above by building without PCH.  However, now I'm faced with the following error:

In file included from /home/chris/Documents/Unreal Projects/Blocks 4.20/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/Blocks/Development/AirSim/Module.AirSim.cpp:2:
In file included from /home/chris/Documents/Unreal Projects/Blocks 4.20/Plugins/AirSim/Source/AirBlueprintLib.cpp:4:
/home/chris/Documents/Unreal Projects/Blocks 4.20/Plugins/AirSim/Source/AirBlueprintLib.h:142:58: error: 
      member access into incomplete type 'UWorld'
        APlayerController* controller = actor->GetWorld()->GetFirstPlayerCont...
                                                         ^
/home/chris/mkdev/UnrealEngine/Engine/Source/Runtime/Core/Public/GenericPlatform/GenericPlatformMisc.h:867:25: note: 
      forward declaration of 'UWorld'
        static bool Exec(class UWorld* InWorld, const TCHAR* Cmd, FOutputDevice& Out)

msb336, 2019-10-03:
I believe this issue is solved in newer versions of the engine (`4.18+`). If you are still encountering the problem, please re-open the issue.

closed 2019-10-03 by msb336