microsoft/AirSim #804: moveToPosition API call ends in offset

SweetShot, 2018-02-09:
Hello,
I am using Python Client API to communicate with drone and currently using move* API. I am having some issues with moveToPosition API call, using simGetPose() in the loop after giving the command to move to a certain position I am noticing certain offset in the location. 
As of now, I am printing out the straight line distance between the target position and the position at which drone stabilizes, it usually ends on ~0.38 (from 0.25 to 0.45) off from target. 
Another thing I have noticed is that if I am calling moveToPosition with a new location that's under 2 units from current location then drone barely moves. 
Also the above is seen if I continuously keep sending moveToPostion with exact same position (in the loop), i.e. drone stabilizes around 1.9 to 2 units apart from the target location.

the code doing this is:-  (this code is inside a while True loop)
`if time.time() - time1 > 10.0:
        # get new random place 
        pos = [rnd.randint(-10, 10), rnd.randint(-10, 10), rnd.randint(-10, -3)]
        time1 = time.time()
        client.moveToPosition(pos[0], pos[1], pos[2], 5, 0)
    currPos = client.simGetPose().position
    
    print( ((pos[0]-currPos.x_val)**2 + (pos[1]-currPos.y_val)**2 + (pos[2]-currPos.z_val)**2)**(1/2) )`

Am I having some issues with the coordinate system of the drone and global coordinates?
Thank you

sytelus, 2018-02-13:
Yes, this is known issue... The problem is that quadrotors don't have breaking system like cars do. So when distances are very small, they need to move with small velocities. This causes longer time to stabilize. So the  compromise is that when distance is within some delta, we stop iterations. 

I certainly think better algorithms are possible. Unfortunately, we don't have bandwidth to improve this particular aspect at the moment. However you are welcome to contribute. The code you want to look at is [this line](https://github.com/Microsoft/AirSim/blob/2bc9f72d9f09f9f33a82cf3744445af07857a32f/AirLib/src/vehicles/multirotor/controllers/DroneControllerBase.cpp#L166) which stops interaction when remaining distance is within breaking_dist.

SweetShot, 2018-02-15:
Got it !! I will certainly have look into that

closed 2021-11-30 by jonyMarino