microsoft/AirSim #806: moveByVelocity using x, y components changes the altitude over time

Mohamed-Kassem, 2018-02-10:
I am running an RL experiment using the drone in the blocks environment. I use the moveByVelocity API call and the control just changes vx, vy (keeping vz=0). However, after sometime the altitude changes, you can try running  this loop for example:
```
print(client.getMultirotorState())
for i in range(10):
  client.moveByVelocity(2,0,0,3)
  time.sleep(5)
  client.moveByVelocity(-2,0,0,3)
  time.sleep(5)
print(client.getMultirotorState())
```

This causes other problems during training, is this a normal behavior? and if so how to avoid it?

Thanks in advance.

sytelus, 2018-02-13:
This is normal. The moveByVelocity only tries to achieve desired velocity along axis. There is no closed control loop for position. For quadrotors, z axis stability (or any axis stability) requires closed loop control because IMU by itself can't tell accurate position. So whenever you use moveByVelocity, you are not guaranteed about stability on any axis. If that's what you want then you might want to try moveBy Position API.

Mohamed-Kassem, 2018-02-13:
Thanks a lot, I will be using moveByPosition as suggested. Does moveByPosition use the GPS for stabilization?

sytelus, 2018-02-13:
Yes, it uses GPS.

closed 2018-02-13 by sytelus