microsoft/AirSim #808: Low level quadcopter control

zbenic, 2018-02-13:
Hello.

My plan is to recreate the work done in:
https://arxiv.org/pdf/1707.05110.pdf

and to continue to research further. The only thing missing is the access to the low level quad controls (to be able to set individual motor rpms or thrusts).

Regarding the closed issue:
https://github.com/Microsoft/AirSim/issues/99

Are there any progress on the said matter?
I am also willing to help with the necessary code changes.

sytelus, 2018-02-13:
It's just matter of adding new API call, for example, `void setActuators(std::vector<float>)`. Then in simple_flight, just set these values directly via `PassthroughController`. Nothing too hard, just bit of plumbing. However as I'd mentioned in [my comment](https://github.com/Microsoft/AirSim/issues/99#issuecomment-285547435), RPC APIs have poor throughput so this might not work...

zbenic, 2018-03-05:
Thanks for the advice.
I've managed to do it.
Created new movebyrotorspeed command which converts input argument that are rotor RPMs to arguments for the PassthroughController (roll, pitch, yaw and throttle control variables).

My next step would be to implement the functionality that is mentioned:
https://github.com/Microsoft/AirSim/issues/600

What I would like to do is get the rotor speeds via the reinforcement learning, apply them by using movebyrotorspeed command, and get the state from the environment. I would like to do that with the fixed timestep dt.

Can you give a few pointers like in your last answer?

nrocketmann, 2018-06-28:
Hi zbenic, I am also trying to reproduce the results from that paper, but I'm very new to airsim and C++ in general. How did you gain access to the low level commands for rpms on each rotor, etc.? What would I need to do to use those?

zbenic, 2018-06-29:
Hello @nrocketmann.

Actually, I've given up on using AirSim for my purposes because I've switched to  https://github.com/wil3/gymfc which should be OpenAI Gym environment for testing RL on quadcopters.

nrocketmann, 2018-06-30:
@zbenic Thanks for the info. I'll give that a try. Good luck!

vinbo, 2018-06-30:
Oh, that sounds a sad news to me @zbenic , how do you compare the two, which part of gymfc do you think is better than airsim?

wil3, 2018-07-04:
Hey @vinbo gymfc author here. Airsim is a simulator while gymfc is an environment for constructing intelligent flight control systems so they have different goals. Currently gymfc has environments for learning attitude control but the goal is to expand this to include navigation and guidance. Gymfc is using Gazebo as the backend physics engine, while airsim is uses the unreal engine. Gymfc could potentially use unreal as its backend, would just need write the interface to the aircraft model to control the actuators and read sensor data. Luckily Gazebo already provides an interface for controlling a quadcopter which made it easiest for gymfc to use this engine.  You wouldnt use gymfc as a flight simulator, the worlds are constructed specifically for learning certain tasks and the graphics aren't nearly as impressive as airsim. I've been working on another project for doing SITL testing of [betaflight](https://github.com/betaflight/betaflight) called [Aeroloop](https://github.com/Aeroloop) which I'd love to port over the backend currently using Gazebo to Airsim.

vinbo, 2018-07-05:
Thank you for your introduction, wil3, already stared. I'll look into it. Thanks

RVBldr, 2018-10-19:
@zbenic Can you confirm that we consider this closed?

zbenic, 2018-10-20:
Yes, we can close this.

juanmed, 2019-01-07:
@zbenic  Hello! Hey I read what you did regarding controlling the rotation speed of the rotors and also briefly saw your changes in your personal repository. Could you please let me know if you were able to actually make the drone move in the simulator via commanding rotor speeds in the way you did it? More strictly, if through your movebyrotorspeed command function, the drone in the simulator actually moved according to forces generated by those rotor speeds? I would like to develop a low level controller based on other control strategies different from the implement PID, so I would need to command rotor speeds and the simulated drone state to change according to those commanded rotor speeds.

I understand you dropped airsim, but just would like to confirm up to which point you were able to get, and maybe I can pick up from there.

Thanks for your answer.

zbenic, 2019-01-08:
@juanmed Hello! Yes, I was able to use the movebyrotorspeed command to move the quad in the simulator. Good luck!

juanmed, 2019-01-08:
Nice to know! Thanks!

linxiaobo110, 2019-03-12:
@zbenic 
Hi, I am do research about reinforcement learning for quadrotor too. I use the DDPG. Now I want to simulate in airsim. . AsI used to code in tensorflow in python, can the low level control be implemented in airsim + python?
And, do you know where can find bbs or community for this subject?

linxiaobo110, 2019-03-12:
> @linxiaobo110 <https://github.com/linxiaobo110>
> Hello. I believe everything is explained here: #1680 (comment)
>
> The low level control is already implemented.

Thank you!
I am reading it now.
Do you still do research with airsim? Can gymfc work with position control?
I am the only one do research about RI on quadrotor. how hard it is!

sincerely, xiaobo

zbenic, 2019-03-13:
I'm not doing anything in AirSim for the moment.
You can use gymfc to simulate problems of position control.

wil3, 2019-03-13:
@linxiaobo110 new version of gymfc will be released in the next couple weeks making it even easier to do trajectory and navigation control. Out of the box the current gymfc version is for attitude control.

zbenic, 2019-03-13:
@wil3 Which version of Gazebo will it use and will it work on Ubuntu 18.04? I remember when setting up Gazebo 8, I had some problems with dependencies on 18.04 so I had to revert to 16.04.

wil3, 2019-03-13:
Hi @zbenic Gymfc is under pretty heavy development and changes fast. Ubuntu 18.04 is now supported and Gazebo 8 and 9. The Gazebo plugins are now built dynamically on install for whatever version you have installed. The new version is basically a complete rewrite though addressing a lot of dependency issues with the aircraft, I'm almost done with it.

zbenic, 2019-03-13:
@wil3 Great, thank you for your efforts. Looking forward to the new version.

linxiaobo110, 2019-03-14:
Thank you very much!!!!
These messages are very helpful for me! You are so nice!!!

closed 2020-09-04 by jonyMarino