microsoft/AirSim #809: Stale sensor factory pointer

cdyk, 2018-02-13:
In  ASimModeWorldMultiRotor::createVehicle,

https://github.com/Microsoft/AirSim/blob/1c07c9bfc6dd087209ee6be4936a10a3e90d2031/Unreal/Plugins/AirSim/Source/Multirotor/SimModeWorldMultiRotor.cpp#L187

An UnrealSensorFactory is created on the stack, and a pointer to this object is passed to createConfig. This pointer is then passed to e.g. the Px4MultiRotor constructor, which stores this pointer. That object is stored in SimModeMultiRotor's vehicle_params_-array (and it is passed into MultiRotorConnector). When createVehicle exits, the sensor-factory goes out of scope, and the pointer in the vehicle_params is stale.

One simple resolution is to new the sensor_factory, wrap it in unique_pointer and move it into the vehicle params.

sytelus, 2018-02-13:
Good catch! Looks like its not easy to avoid shared_ptr. One issue is that sensor factory is dependent for each pawn (because ray cast needs to ignore that pawn for collision).

Just curious, how did you found this issue?

closed 2018-02-13 by sytelus

cdyk, 2018-02-13:
Thanks!

If you manage to tie ownership to something (e.g. ASimModeWorldMultiRotor) you should get away with a unique_ptr and avoid the cost of shared_ptr multithreaded reference counting...

It popped up when I re-implement the recent changes in the Unreal glue code for our rendering engine. As this engine is organized differently and is threaded differently, I have to read parts of the code very carefully. :-)

sytelus, 2018-02-17:
This is few places where I think shared_ptr makes sense. Basically, we might want to have factory available to multiple objects so they can use it as later time. So its shared ownership.