microsoft/AirSim #810: MavLinkDrone

cdyk, 2018-02-13:
In
https://github.com/Microsoft/AirSim/blob/1c07c9bfc6dd087209ee6be4936a10a3e90d2031/AirLib/include/vehicles/multirotor/controllers/MavLinkDroneController.hpp#L768
when invoked at
https://github.com/Microsoft/AirSim/blob/1c07c9bfc6dd087209ee6be4936a10a3e90d2031/AirLib/include/vehicles/multirotor/controllers/MavLinkDroneController.hpp#L790
uses a nullpointer if one uses the airlib-SensorFactory, as the switch case
https://github.com/Microsoft/AirSim/blob/1c07c9bfc6dd087209ee6be4936a10a3e90d2031/AirLib/include/sensors/SensorFactory.hpp#L21
doesn't handle SensorType::Distance. The UnrealSensorFactory handles this sensortype, so under Unreal there is no problem.

If msr::airlib::SensorFactory is intended to be instantiated, it should either handle all SensorTypes, or MavLinkDroneController must somehow handle null-pointers (at least assert).

sytelus, 2018-02-13:
Good catch again. A simple check now avoids using unsupported sensor by sensor factory... Not yet sure if I should raise exception here. I don't have PX4 setup at the moment however so couldn't test that part. If you still see the issue, please let me know.

closed 2018-02-13 by sytelus

cdyk, 2018-02-13:
Thanks!

If using a null distance sensor (or a distance sensor that always returns max distance) makes sense, I would make DistanceSimple::getRayLength concrete and returning maxDist, and return this implementation in SensorFactory::createSensor.

Otherwise, if it doesn't make sense, and the 'platform' must somehow provide this result, making the constructor of SensorFactory private makes that class basically abstract so compilation will fail when using SensorBase directly.