microsoft/AirSim #811: Error while running hello_car.py

PCD873, 2018-02-13:
I just started to experiment AirSim python API on the City environment and I am getting an error when trying to run hello_car.py : 

```
Traceback (most recent call last):

  File "hello_car.py", line 18, in <module>
    car_state = client.getCarState()
  File "AirSimClient.py", line 563, in getCarState
    return CarState.from_msgpack(state_raw)
  File "AirSimClient.py", line 33, in from_msgpack
    obj.__dict__ = { k : (v if not isinstance(v, dict) else getattr(getattr(obj, k).__class__, "from_msgpack")(v)) for k, v in encoded.items()}
  File "AirSimClient.py", line 33, in <dictcomp>
    obj.__dict__ = { k : (v if not isinstance(v, dict) else getattr(getattr(obj, k).__class__, "from_msgpack")(v)) for k, v in encoded.items()}

 AttributeError: 'CarState' object has no attribute 'position'
```

I tried to build it from source with the blocks environment and it worked fine. I looked around and my best guess it that this happens because the API changed since the binaries for AirSim were released, so the getCarState() function is now different. 
Am I missing something ?

If my guess is right, are you planning on releasing new binaries with the updated API ? If not, could you point me toward a commit in which the getCarState() would work with the latest release of the binaries ?

sytelus, 2018-02-15:
Is this happening with latest binaries and latest AirSim client code?

PCD873, 2018-02-15:
Yes, I am using the commit from today (just got commit  714c265) and the error is still there. I use the latest release of the binaries : v1.1.7, released this on 1 Dec 2017 (commit 39aae01 I believe).

I am running on Windows 7 Windows 7 Professional.

micmancini16, 2018-02-15:
I have the same issue on the Neighbourhood binaries. CityEnvironment binaries works fine. Both binaries are from the 1.1.7 release.

sytelus, 2018-02-16:
ok... I created the new release 1.1.8. Please give it a try and let us know if this issue still persist.

micmancini16, 2018-02-16:
I have tested the 1.1.8 release for both Neighbourhood and CityEnvironment binaries. The API seems to work fine on both environments, thank you! 

Unrelated to the API issue, the 1.1.8 version of the CityEnvironment lags a lot and run with a much lower FPS rate compared to the 1.1.7 version ( 40 FPS on 1.1.7, about 7-8 FPS on 1.1.8). Neighbourhood binary runs fine on both releases.

sytelus, 2018-02-17:
CityEnviron will require beefy GPU. We have been trying to optimize but still ways to go. You might want to start exe as lower resolution,

MyGame.exe  -ResX=640 -ResY=480 -windowed

closed 2018-02-17 by sytelus