microsoft/AirSim #815: Car client.reset() does not reset speed. Ready to work on a Pull Request but need some advice.

miev88, 2018-02-14:
I'm doing reinforcement learning and I'd like to reset an episode after a collision. I'm using client.reset() from the API in PythonClient but this call does not reset speed after a crash unfortunately, and at the beginning of the following episode the car is still moving (you can observe this behavior in "Reinforcement Learning for Car Using AirSim" that @sytelus posted on YouTube too). In particular, speed gets stuck with a negative number after a collision in the Neighborhood (binaries), whereas speed takes the value of the last speed registered before a crash in the City (binaries). I have written a script in Python that solves this problem, but it is a quite dirty solution. 

I would like to fix the problem "at the source". I had a look at the AirSimClientBase class where the method reset() is implemented. There is a call to another reset method, which (I assume) is implemented in C++. In RpcLibClientBase.cpp the method reset() is implemented with a pointer to implementation (pImpl). However, I cannot understand where this implementation is. I have no experience with C++,
but I'm ready to invest some time to fix the problem if someone can point me in the right direction.

sytelus, 2018-03-02:
I just tried out few things and added enhancement that now stops car immediately on reset(). However RPM still keeps rolling for a while although I don't see any effect due to that. Please give it a try to latest version.

Related commit: https://github.com/Microsoft/AirSim/commit/695d8acf51c515628288b369c8812025d7b96a1d

miev88, 2018-03-13:
I've pulled the repo and downloaded the latest release for Neighborhood (v 1.1.8 - Windows). reset() resets the position, but not the speed.

That's the code I've used for testing in Windows 10 with Python 3.6:

```
import time
from AirSimClient import *

client = CarClient()
client.confirmConnection()
client.enableApiControl(True)
car_controls = CarControls()

car_controls.throttle = 1
client.setCarControls(car_controls)
time.sleep(15)

client.reset()
client.enableApiControl(False)
```

melange396, 2018-07-11:
this does the trick for me! ^

closed 2020-09-01 by jonyMarino