microsoft/AirSim #817: Trouble using the moveByManual() function from the MultiRotor-api

Swim0n, 2018-02-14:
Hi there.

I am currently working on a project controlling a drone using a neural network trained on image data and RCData.

I have implemented getRCData() and setRCData() in the AirSimClient API (I'm using Python), and verified that this is working by observing the data being correctly set in the SimpleFlightDroneController, but I'm running into problems trying to use the moveByManual() function. What I gather from the API-description and from looking at the actual C++ code for the function in DroneControllerBase is that previously set RCData should be read and then used to control the movement of the drone, but whatever parameters i pass I can't seem to get any movement at all.

```
client = MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

client.takeoff()
client.moveByManual(10, 10, 0, 60)

while (True):
    stamp = client.getMultirotorState().timestamp
    client.setRCData(rcdata = RCData(timestamp = stamp, throttle = 1.0, is_initialized = True, is_valid = True))
```

This is the code I'm currently using. I have also tried putting the moveByManual()-call in the loop, but to no avail. I'm wondering if anyone has any insight as to why this is not giving any movement at all? Is there something that flies over my head regarding the API-calls?

Any help is greatly appreciated!

sytelus, 2018-02-15:
Could you submit pull request so we can see your implementation. Why it works or doesn't depends on how you have implemented it.

Swim0n, 2018-02-15:
Thanks for your answer Shital. I submitted a pull request just now. Thanks so much for your help!

Swim0n, 2018-02-15:
I wanted to add that I got a little bit further in my troubleshooting. I found that something is consistently resetting the RCData to 0-values and false boolean values.

```
client.setRCData(rcdata = RCData(timestamp = stamp, throttle = 1.0, is_initialized = True, is_valid = True))
print(client.getRCData())
client.moveByManual(10, 10, 0, 60)
time.sleep(2)
print(client.getRCData())
```
The code above prints the correct RCData (that I just set) with the first print, but the second print only gives 0-values. Does anyone have any clue as to what might be happening?

Cheers,

cangokalp, 2018-02-15:
@Swim0n this would be a great feature. So with this - you can have your neural net output roll, pitch , yaw, throttle values and pass them to RCdata, is this right?

Swim0n, 2018-02-15:
@cangokalp yes, that is the idea. I am going to train the network in AirSim and then apply it to a real-life drone, so it's important to be able to work with RCData.

I've moved a bit further in my debuggning now and it seems like there were two reasons as to why nothing happened when setting RCData. 

The first one was that when the MultiRotorConnector updates the rendered state it also sets the RCData of the flight controller to whatever input it gets from the Joystick(which is nothing), if the RemoteControlID of the controller was equal to or higher than 0, which I solved by setting the default RemoteControlID of the controller to -1.

The second one was that the moveByManual-message from the API took to long to reach its implementation in DroneControllerBase, which lead to the age of the message being higher than the maximum allowed message age, which I solved by setting the kMaxMessageAge to an arbitrarily high value.

I have also added the fixes above to my pull request.

Now, I can get movement from the API-calls, but it's still VERY wonky. I am using this code:

```
client.setRCData(rcdata = RCData(pitch = 0.0, throttle = 1.0, is_initialized = True, is_valid = True))
client.moveByManual(vx_max = 10, vy_max = 10, z_min = 0, duration = 1000000)
```
and running it once doesn't do anything. BUT, if I run it a second time without restaring the simulation, it pops off into the air!

I would deeply appreciate some insight from the developers regarding how all these functions interact with each other and how I should move forward. Thank you all very much in advance!

cangokalp, 2018-02-15:
What do you mean by wonky? Like it moves weirdly? or sometimes works sometimes doesn't?
Also - Did you try it with z_min=10. From what I understand looking at the moveByManual() code (contrary to the documentation) z_min is (not exactly) pretty much vz_max.

Swim0n, 2018-02-15:
No, it's wonky that I have to run the script twice to get any movement. And I have tried a bunch of different values but it doesn't change anything. It moves upwards really fast so I don't think the z_min is a problem here.

sytelus, 2018-02-16:
I found the cause. moveByManual tried to estimate RC trim and it had validation to check for number of RC messages received for trim estimation which failed.

Also, you want to first call moveByManual and the set RCData.

I'm still not exactly sure what you want to use moveByManual. The moveByManual basically interprets (pitch, roll, thrust) as (dx, dy, dz) in initial body frame. This method was initially implemented to achieve so called "headless mode" for RC. This mode basically means you are moving quadrotor with reference to you and so you don't have to care where it's exactly pointing. For example, when quadrotor is very high up, you can't tell which direction front is pointing to. You just want to send command to move forward where forward means in the direction where it was initially pointing to before we did take off. Also, it has these min and max restrictions to implement what we called "toddler mode". The idea is that you can give RC to a toddler and he might do whatever random thing but we want to make sure that quadrotor doesn't move out of a virtual box area in space and never goes faster than specified velocity. Also, as toddlers have hard time to figure out where drone is pointing to now, the "headless RC" mode is more intuitive for them.

If you just want vehicle raw command then you should use moveByAngle instead.

I've merged your PR and also made above fixes so things are not "wonky" any more I also added sample py script to use this mode. See this commit: https://github.com/Microsoft/AirSim/commit/527916c89a53b9569e1d7f644d3f385202151a73

Swim0n, 2018-02-16:
Thank you so much for your response Shital. I understand now what the moveByManual() is supposed to achieve and is indeed not the functionality I am looking for.

Actually I decided on a different approach, not enabling API control but still setting the RCData. This works perfectly since not enabling API control allows the program to update the motor thrusts like the RCData was coming from a joystick or some other controller.

Thanks alot for your help and cheers for merging my PR :)

closed 2018-02-16 by Swim0n

cangokalp, 2018-02-16:
I want to give roll, pitch, yaw, throttle commands to the drone in my reinforcement learning algorithm - and I thought moveByManual exactly does that, like as if you have the controller and moving the sticks - doesn't it? 

Controlling via moveByAngle doesn't allow to set throttle. I want to mimic a controller pretty much, so I want to control with roll pitch yaw throttle as if I'm holding a controller and moving the sticks - except reinforcement learning algorithm will set those values and find out what values to set each time step.

@Swim0n how do you set the RCData now? Can you send another PR :) ?

cangokalp, 2018-02-16:
When I run the sample py script, drone doesn't move.

sytelus, 2018-02-17:
moveByAngle takes z as parameter which automatically adjusts throttle. For RL, you might want to assume that PID controller is already in place which takes higher level commands. 

`def moveByAngle(self, pitch, roll, z, yaw, duration)`

It is possible and fairly easy to add new API for angle + thrust instead of angle + z. If you need it, please feel free to contribute (see PR by @Swim0n for example).