microsoft/AirSim #822: Please stop using Github

azvampyre, 2018-02-16:
I have tried downloading the binaries 10 times, the zip files always fail and Github does not support file resuming. Quit using this garbage site and post them on MSDN, thanks.

sytelus, 2018-02-16:
Where are you located? I haven't seen this problem in US or Europe at least.

azvampyre, 2018-02-16:
Tonopah, Arizona (Belmont)
Try downloading, and pause it, you'll see it will not resume. Links to open source code should never be encrypted.

azvampyre, 2018-02-16:
The problem appears to be related to AmazonAWS. I resorted to tracking the source to find a direct link to the AmazonAWS storage and download it from there using jdownloader. This link is ridiculous though, no wonder it would not resume, and note that the Expires time is only 300 seconds. How is someone supposed to download 1.2GB in 300 seconds? I had to change that to 999999 because the files would complete at 300 seconds with corrupted EOF. Thanks for the help nonetheless.

https://github-production-release-asset-2e65be.s3.amazonaws.com/81887991/afc178d0-d635-11e7-80e7-f03ca1201690?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180216%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180216T064741Z&X-Amz-Expires=300&X-Amz-Signature=d61e3d4eb69fe7b1f41b22869ee4a80272674848ea28fa94a25af7028ff05623&X-Amz-SignedHeaders=host&actor_id=11075597&response-content-disposition=attachment%3B%20filename%3DLandscapeMountains.zip&response-content-type=application%2Foctet-stream

n1ckfg, 2018-02-23:
I don't think this is a problem with GitHub necessarily--I see this issue on Windows, but not on Ubuntu.

sytelus, 2018-03-15:
This is also a problem in Chrome. Not being able to resume downloads seems to happen frequently in Chrome. Just found this app [uGet](https://sourceforge.net/projects/urlget/?source=directory) which has ability to resume this downloads. It also allows to use multiple connections so if you are on the other side of the world, download can be speeded up by 5X by using 5 simultaneous connections.

sytelus, 2018-04-12:
There is easier and better solution I found: just use download manager like uGet: http://ugetdm.com/. You can also turn on multiple thread which speeds up things significantly.

closed 2018-04-12 by sytelus

kk99351, 2019-12-19:
use multicloud manger(multicloud.com) to transfer files directly to your cloud storage like gdrive,mega,etc and then download it from your cloud storage.

nobicycle, 2022-09-06:
> use multicloud manger(multicloud.com) to transfer files directly to your cloud storage like gdrive,mega,etc and then download it from your cloud storage.

Github is not supported by multicloud
https://www.multcloud.com/support-clouds/